realthunder/FreeCAD #816: Errors produced by Linkmarge

wsteffe, 2023-05-16:
Following errors were generated by LinkMerge after changing a few parameters of annexed document: 

```
Merging tempSheet.B2 to Variables.B2 --> 8.385
Merging tempSheet.B4 to Variables.B4 --> 3.19
Merging tempSheet.B7 to Variables.B7 --> 1.11
Merging tempSheet.B8 to Variables.B8 --> 1.51
Merging tempSheet.B9 to Variables.B9 --> 2.563
Done merging tempSheet into Variables
Cells changed = 6
Sketch Sketch001 has partially redundant constraints!
<App> FCvarUpdate.py(42)|PropertyLinks.cpp(439): CMP_KaCh2_SFILT_6_4_wg_tunable_V3#Sketch007.Support missing element reference CMP_KaCh2_SFILT_6_4_wg_tunable_V3#Import004 ;#7813;:H12f6,F;:H12d7,F.Face98
2.2e-08 <App> FCvarUpdate.py(42)|Document.cpp(4203): Failed to recompute CMP_KaCh2_SFILT_6_4_wg_tunable_V3#Sketch007: AttachEngine3D: subshape not found Import004.Face98
Recompute failed! Please check report view.
0.0691592 <App> Document.cpp(4203): Failed to recompute CMP_KaCh2_SFILT_6_4_wg_tunable_V3#Sketch007: AttachEngine3D: subshape not found Import004.Face98
Recompute failed! Please check report view.

[CMP_KaCh2_SFILT_6_4_wg_tunable_V3.zip](https://github.com/realthunder/FreeCAD/files/11485123/CMP_KaCh2_SFILT_6_4_wg_tunable_V3.zip)

```
To change the variables I am using the annexed macro which merges the values defined in annexed .csv file. But you may also change them directly in the Variables Spreadsheet.

Recomputation is sucessfull with LinkDaily 
[FCvarUpdate.zip](https://github.com/realthunder/FreeCAD/files/11485156/FCvarUpdate.zip)

realthunder, 2023-05-20:
Can you please post  your document here? There is some changes in topo naming encoding. Please do a full recompute first. Right click the document icon and choose `Mark to recompute`.

wsteffe, 2023-05-20:
Hi RT, I was thinking to have already uploaded it but here it is:
[CMP_KaCh2_SFILT_6_4_wg_tunable_V3.zip](https://github.com/realthunder/FreeCAD/files/11522132/CMP_KaCh2_SFILT_6_4_wg_tunable_V3.zip)

Now I tried also the full recompute as you suggested but it is still failing after a variable change. It is sufficient a small change in the last digit of any variable.

commit c777daa4 (2023-05-21): PD: fix pocket profile normal direction

Fixes #816
@wsteffe
closed 2023-05-21 by wsteffe