realthunder/FreeCAD #818: Build - FAILED: src/Base/CMakeFiles/FreeCADBase.dir/Base64.cpp.o - needs #include <cstdint>

thx1111, 2023-05-16:
Arch linux
linux 6.2.12.arch1-1
clang 15.0.7-9

Name            : freecad-linkstage3-git
Version         : Tip.r1.g5896ded8d7-1

This not the same build error fixed for `src/App/ComplexGeoData.cpp`.

```
In file included from /var/build/freecad-linkstage3-git/src/FreeCAD/src/Base/Base64.cpp:37:
/var/build/freecad-linkstage3-git/src/FreeCAD/src/Base/Base64.h:314:9: error: no type named 'uint8_t' in namespace 'std'; did you mean simply 'uint8_t'?
```
See at https://en.cppreference.com/w/cpp/header/cstdint

Add to Base64.cpp
```
#include <cstdint>
```

realthunder, 2023-05-20:
Not sure if you are aware, the new `Tip` release is from [LinkMerge](..//tree/LinkMerge) branch. It is still in early test. And I plan to do another merge with upstream in near future.

closed 2023-05-20 by realthunder

thx1111, 2023-05-20:
No, didn't know that.  Thanks!

commit 36afaca6 (2023-05-20): Fix build error

Fixes #818