realthunder/FreeCAD #821: Render Cache not work, slow FPS in LinkTip release

ftdt10, 2023-05-18:
Set the Render Cache to Experimental, 
but frames per second is decreased by half in LinkTip release version.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.516.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: b3832bb2e9026abbb57a2f726603fa97994bc930
Python 3.11.3, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Curves 0.6.9
  * fasteners 0.4.56
  * kicadStepUpMod 10.19.1
  * kicadStepUpMod.backup1684338274.0809298 10.19.1 (Disabled)
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Render 2023.2.5
  * Render.backup1684338330.7047546 2023.2.4 (Disabled)
  * sheetmetal 0.2.63
  * TabBar
[/code]

realthunder, 2023-05-20:
I can't seem to reproduce the problem. Any particular file you are testing? Are there any frame rate difference in render cache `Experimental` and `Auto`.

ftdt10, 2023-05-21:
The problem only happens on Windows.
It's normal on Ubuntu.

I renamed bin/opengl32sw.dll to bin/opengl32.dll
due to black screen problem.
https://forum.freecad.org/viewtopic.php?t=36087&start=40
But, I think it has no relation with the probelm.

realthunder, 2023-05-23:
When do you have this message? When FreeCAD just start up? What graphics card do you have?

ftdt10, 2023-05-23:
I just tested in other PC with AMD 5600G iGPU or NVidia GTX1650 dGPU, 
but there is no black screen problem.
It seems it's problem for the older graphics card.
I hope it works for the HD5500 also.

realthunder, 2023-05-24:
Can you please try download again. Same name, but rebuilt with older qt. Let me know if it works.

ftdt10, 2023-05-24:
It works now. 
There is no black screen error. The FPS is at its maximum as before.
Thanks for the rebuild.

closed 2024-02-18 by ftdt10