realthunder/FreeCAD #823: Build - LinkMerge - src/FreeCAD/src/3rdParty/libE57Format/include/E57Format.h - also needs #include <cstdint>

thx1111, 2023-05-20:
Arch Linux

Now in the LinkMerge branch, same as before:

```
FAILED: src/3rdParty/libE57Format/CMakeFiles/E57Format.dir/src/BlobNodeImpl.cpp.o
/var/build/freecad-linkstage3-git/src/FreeCAD/src/3rdParty/libE57Format/include/E57Format.h:42:10: error: no member named 'int16_t' in namespace 'std'; did you mean '::int16_t'?
...
```
needs `#include <cstdint>`

realthunder, 2023-05-20:
Can you please do a `git pull`? I think your local `LinkMerge` branch is not up to date. And also, please clear your build directory (or make a new one) to do a clean build to avoid mixing with `LinkDaily`.

thx1111, 2023-05-20:
Oops - sorry - `pacman` is not quite as automated as I had imagined.  I'm doing a complete cleanbuild now.

thx1111, 2023-05-21:
Ok, there's just this one that still needs an `#include <cstdint>`:
`src/FreeCAD/src/3rdParty/libE57Format/include/E57Format.h`
There were no other build issues.

thx1111, 2023-05-21:
Thanks Lei