realthunder/FreeCAD #824: New 2023-05-16 appimage does not use existing configuration path?

nathaneltitane, 2023-05-20:
I have pre-existings configurations and prefs that used to be loaded from ~/.FreeCAD and for some reason, even though they are now located under ~/.config/FreeCAD, the appimage (extracted or not) does not seem to read from any of those directories?

What changed? Why arent't the user preferences being read?

realthunder, 2023-05-21:
The upstream has changed the configuration directory to `~/.config/FreeCAD`, and macro/addon to `~/.local/share/FreeCAD`. You either copy/link the old files to the new location, or start the application with command switch `--keep-deprecated-paths`.

nathaneltitane, 2023-05-21:
i did in fact copy them and they are still not being read...

realthunder, 2023-05-21:
I can't reproduce it here. Can you please elaborate your process? When you change some setting in FreeCAD and close the application, are there any updates in `~/.config/FreeCAD/user.cfg` or `~/.FreeCAD/user.cfg`?

nathaneltitane, 2023-05-25:
> I can't reproduce it here. Can you please elaborate your process? When you change some setting in FreeCAD and close the application, are there any updates in `~/.config/FreeCAD/user.cfg` or `~/.FreeCAD/user.cfg`?

I'd have to double check when back home: essentially i have link.system.cfg link.user.cfg system.cfg and user.cfg under ~/.FreeCAD for now with the previous link release. I had copied them over to ~/.config/FreeCAD and nothing happened when launching the new version essentially.

wasteoinc, 2023-06-16:
There is an issue but there is also a nice workaround
https://github.com/realthunder/FreeCAD/issues/833#issuecomment-1594277660

commit 47ae0730 (2023-06-18): App: fix handling of vendor prefixed user config

Related #833
Fixes #824