realthunder/FreeCAD #827: circle diameter of projected External Geometry is not exact, in Link branch release.

ftdt10, 2023-05-25:
Uploaded [test.zip](https://github.com/realthunder/FreeCAD/files/11560642/test.zip) (test.fcstd) content is below.
1. Body > Pad > Sketch, the Sketch content is a rectangle and 4.3mm diameter circles
2. Body > Sketch001, with the Sketch001 Attachment at the Pad side face
3. in the Sketch001, add External Geometry of the circle edge

But, I found that the circle edge length is 4.25302mm, not 4.3mm.
In FreeCAD upstream release, it is correct 4.3mm.

wohltat, 2023-05-25:
I can reproduce this. Wrong dimensions appear also when referencing the pads original sketch directly and also if XZ-plane is used instead of pad side face.

The dimensions change a little if the sketch plane is changed. If the sketch plane is parallel to the circles plane then the dimensions are correct.

-------

[code]
OS: Linux Mint 20.1 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 2023.516.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: b3832bb2e9026abbb57a2f726603fa97994bc930
Python 3.11.3, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
[/code]
anch: HEAD
Hash: b3832bb2e9026abbb57a2f726603fa97994bc930
Python 3.11.3, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
[/code]

3xxx, 2023-08-01:
https://github.com/realthunder/FreeCAD/issues/836

commit 83ace59e (2023-08-06): Sketcher: use analytical solution when projecting circle to line

Fixes #827
Fixes #836