realthunder/FreeCAD #839: Transform Tools Can't Select Multiple Features

Trakyan, 2023-06-24:
So I noticed this issue in the mirror tool intitially, but it seems to occur with the other transform tools (linear pattern, etc.)
The tool "works" but only on the last feature before the transform. Ticking the box beside other features does nothing, and when you click OK in the combo view to close the transform, and open it again all the boxes are unticked. 

This issue seems to exist in the Tip release but not the 2023.01.31 version from what I can tell. 

If a mirror is created with multiple features in 23.01.31, it will open correctly in the tip release but I won't be able to change what features are used in the mirror.

I'm running EndavourOS which is based on Arch Linux.

Wondering if anyone else can replicate this issue, or if it's again something caused by having mainline and linkstage alongside each other.

Below is a screenshot of the issue.
[image]
[image]

Trakyan, 2023-06-24:
Just tried it on a live USB with only the LS3 appimage, the issue is still there. It is also present in the 0.21 weekly build app image from the main branch (03/06/2023), but not the latest weekly build.

Trakyan, 2023-06-29:
Further testing on a windows laptop has the issue present as well. Someone suggested I try selecting the features from the properties view, and that worked.
Below shows the original bug:
[image]
Then going through the properties panel allows me to actually select multiple (and also not just the most recent) features
[image]

oerkel47, 2023-07-23:
I can confirm this bug and the workaround using the properties panel. Also can confirm the multiple texts on top of each other in the mirror parameter screen.

wohltat, 2023-07-30:
I can confirm the bug. 
Present in 2023-05-21 and not in 2023-01-31.

It looks like this:
[image]

and should look like this:
[image]

In my opinion a mayor bug since the workaround does not offer a preview and almost no user will find it.
Is it just my feeling or is there not much happening at the moment on that branch?

commit 98540849 (2023-07-31): PD: fix auto recompute when editing pattern features

Fixes #839
closed 2023-07-31 by Trakyan