realthunder/FreeCAD #841: Symmetri constrain

ljungis, 2023-06-26:
[image]
 
Allways when adding a symmetri constrain the sketch becomes overdefined.

Any solution?

wsteffe, 2023-06-26:
You have to remove the redundant horizontal condstraint to avoid that.

ljungis, 2023-06-26:
[image]
 
Thank you for your tips.
This is a new behavior to me. The earlier version did not behaive like this.
It must be possible to combine the symmetri constraint with other constraints.