realthunder/FreeCAD #844: BUG: the Addon Manager does not show any indication during installation of Assembly4; and finally the installation fails

raindropsfromsky, 2023-07-05:
I faced this issue while installing the **Assembly4** addon.

After pressing the **Install** button, I see no reaction at all:
- There is no hourglass icon to indicate "wait..."
- The **Install** button does not gray out.
- After the installation is over, the Addon Manager does not show **"Restart now to complete installation"** button.
Thus, there is absolutely no indication that the installation has happened. **[BUG]**

But when I return to the main menu, it shows "Pending restart" against Assembly 4.
So I closed FreeCAD and launched it again.

When I reopened the Addon Manager, it shows that the Assembly4 addon has not got installed. **[BUG]**

Thus, I cannot install Assembly4 at all.

raindropsfromsky, 2023-07-05:
**Update:** The same problem is observed in installation of Curved Shapes Workbench also.

raindropsfromsky, 2023-07-06:
Update: While trying to install **boltsfc**, I saw the following error in the error log:

OSError: [Errno 22] Invalid argument : ' C : /Users/naray/AppData/Locat/cache\\FreeCAD\\Cache\\AddonManager\\MacroIcons\\120px—FCBmpImportScreenshot . '

There were more error lines, but I could not copy them here, because The Addon Manager has a Modal window: It does not allow me to switch over to the error Log pane, which is part of the main GUI.

But as soon as I close the Addon Manager, the Error Log pane is automatically closed. 

Thus, I am not able to scroll through the error log and copy all the messages here.

So I had to use a trick: I kept the Addon Manager on, and then used a software called "Sharex" to OCR the error line from the Error Log pane. But I cannot capture all the error lines with this trick. Please let me know if I can access the error log any other way.

raindropsfromsky, 2023-07-08:
**Update:**

I tried the FreeCAD 20.2, and found that its Addon Manager works perfectly:
1. If the download is taking time, it shows "Downloading xxx MB of yyy total" in the Status Bar of the Download Manager itself.
2. After installation, it pops up a text message, "Successfully installed <addon name><version>".
3. A dialog pops up to ask the user whether to restart FreeCAD, and restarts the app if selects "OK"
4. In case of some addons, another dialog pops up to ask whether the user wants to install a toolbar icon for this addon.

Thus the user is guided completely by the Addon Manager.

All these features are missing from the Addon Manager in the RealThunder fork.
Probably these features were added later in the upstream FreeCAD. (??)

Footz355, 2023-07-17:
I also have problem with TIP release and addon manager. trying to install Sheet Metal WB - nothing happens. When I click Install again I only get: "Cannot launch a new installer until the previous one has finished."- but nbothing seems to install.