realthunder/FreeCAD #848: Mirror feature does not allow adding new features to the mirror after initial creation (tip release)

eblanshey, 2023-07-24:
I am using the latest tip release, Tip 2023.05.21

To reproduce:
1. Create a body above the XY plane with multiple features
2. Mirror the body over the XY plane, and save
3. "Set body tip" to the last feature before the mirror
4. Add a pocket to the body somewhere, save
5. Edit mirror feature, add a "check" to the new pocket
6. Preview does not include the new pocket, and saving the mirror does not show the new pocket on the mirror side

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2023.524.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: f428979aa87456d1ce57e2ea78ee9d70dc7595e4
Python 3.11.3, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * AirPlaneDesign 0.4.0
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * CurvedShapes 1.0.4
  * fasteners 0.4.56
  * Help 1.0.3
  * Plot 2022.4.17
  * QuickMeasure 2022.10.28
```

Trakyan, 2023-07-30:
I opened an issue about this a few weeks ago here
https://github.com/realthunder/FreeCAD/issues/839
It's still present in todays release but in the issue above I mentioned a work around to add/remove features to the mirror/pattern operations.

eblanshey, 2023-08-02:
Thanks, @Trakyan. Fixed in Tip 2023.07.30.

closed 2023-08-02 by eblanshey