realthunder/FreeCAD #855: Position of navigation cube can't be changed

thomasgi1, 2023-07-31:
In MacOS build of Tip 2023.07.30 the position of the navigation cube is always the upper left corner. When I try to select another corner in Preferences/Display/Navigation/Corner, then the related dropdown box is empty, so that no value can be selected.

Zwaubel, 2023-08-08:
Can confirm… this bug still exists in the newest release, even though the release notes mention a fix.

thomasgi1, 2023-08-08:
The bug is partially fixed only. Now, the location of the navigation cube seems to be read from the configuration files, i.e., my original setting of the navigation cube location is working again. But the dropdown box on the settings page Preferences/Display/Navigation/Corner is still empty and can't be used the change the configuration.
[image]

realthunder, 2023-08-09:
Just checked. Yes indeed still got problem. It is probably the resource editor undid my manual changes when saving the file. Will be fixed in next release.

commit ca11b25d (2023-08-06): Gui: fix navigation cube corner setting

Fixes #855