realthunder/FreeCAD #856: BoundingBox not showing

kotrans, 2023-08-01:
Hi
If I turn the Bounding Box option to true (Property) nothing changes. It should show the rectangle and the measurements.

Thanks for your help.
Markus

[image]
[image]

I use:
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2023.731.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 311c0066f7d2e5599a76337706627b6c27f6dc2b
Python 3.11.4, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Installed mods: 
  * boltsfc 2022.11.5
  * Curves 0.6.6
  * DynamicData 2.46.0
  * fasteners 0.4.25
  * Manipulator 1.4.9
  * ThreadProfile 1.85.0
[/code]

Testfile: [BoundingBox.zip](https://github.com/realthunder/FreeCAD/files/12225765/BoundingBox.zip)

eblanshey, 2023-08-04:
Your model is showing a tiny bounding box, but not around the whole part for some reason:

[image]

The bounding box is working for my other parts, though. Maybe it has something to do with your fusion, or something else with the part.

It's working in stable 0.20.2, though. So maybe it's a bug with RT FreeCAD.

[image]

kotrans, 2023-08-07:
Thanks, with version 20230806 it ist working.

commit cc7a1d4c (2023-08-06): Renderer: add support to some legacy coin shape nodes

Fixes #856
closed 2023-08-07 by kotrans