realthunder/FreeCAD #858: Material Young's module is not read into FreeCad

vovodroid, 2023-08-01:
In latest

 [code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2023.731.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 311c0066f7d2e5599a76337706627b6c27f6dc2b
Python 3.11.4, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/Israel (en_IL)
[/code]

version all material Young's module is not read into FreeCad, causing FEM analysis to fault:

[image]

vovodroid, 2023-08-01:
Also it can't be edited in place - each entered value reset to zero.

Neubo, 2023-08-02:
Hi,

I have the same issue.
I get this message:

> 16:27:22  Previously used material card can not be found in material directories. Add document material.
> 16:27:22  Material card chosen:
>     _document_material
> 16:27:22  **YoungsModulus in material data seems to have no unit or a wrong unit (reset the value): Wood**
> 16:27:22  ThermalConductivity not found in Wood
> 16:27:22  ThermalExpansionCoefficient not found in Wood
> 16:27:22  SpecificHeat not found in Wood

```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2023.731.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 311c0066f7d2e5599a76337706627b6c27f6dc2b
Python 3.11.4, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * DVPScaled.brep
  * Manipulator 1.5.0
  * parts_library
  * QuickMeasure 2022.10.28
```

Neubo, 2023-08-07:
Hello,

Tested with latest TIP release: it works now.
@vovodroid, I would let  close the issue if you confirm.

vovodroid, 2023-08-07:
Fixed in TIP 20230806

```
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2023.806.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 93dae74dba4a80dba52f9c5fe73fe5a874bae809
Python 3.11.4, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English
```

closed 2023-08-07 by vovodroid