realthunder/FreeCAD #862: FEM error: wrong ccx binary

Neubo, 2023-08-07:
Hello,

when I double click on "SolverCcxTools,", I get an error message:

[image]

and later, the message in console : 

`CalculiX can not be started. Missing or incorrect CalculiX binary: C:/_Data/Freecad/FreeCAD/bin/ccx.exe`

I've checked the parameters:

[image]

and the file does exist at the right place (same file as in the main Freecad branch)

May you need more information, don't hesitate to ask.

```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2023.806.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 93dae74dba4a80dba52f9c5fe73fe5a874bae809
Python 3.11.4, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * DVPScaled.brep
  * Manipulator 1.5.0
  * parts_library
  * QuickMeasure 2022.10.28
```

Neubo, 2023-08-07:
another thought, I've copied the ccx.exe file from official Freecad to RT Freecad bin directory, and change settings in FEM preferrences. It doeasn't work.
I believe there somthing missing (dll, or other flags) in RT release ?

Neubo, 2023-08-09:
[vovodroid](https://github.com/vovodroid), I just wonder if you don't have the same issue ?

Miksu80, 2023-09-20:
I have same issues with 11.8.2023 and previous and also on linkstage 3

Neubo, 2023-09-20:
Hello @Miksu80, if I'm right, realthunder referenced this issue, but has not yet release it in the last TIP.
But I've heard next TIP is about to come ;)
Maybe a new merge with the latest Freecad mainbranch version ?

Miksu80, 2023-09-20:
@Neubo Yes, the fix didnt make it to the latest tip release. I don't use FEM frequently but was sad to see it failing to start. Hopefully the merge with main branch will happen :)

Neubo, 2023-10-19:
Hello !

It does not work much better with last TIP release (2023-10-18).

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2023.1018.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: ea554afc8427b011877bb340a0f6fd2bd942fe6f
Python 3.11.6, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * DVPScaled.brep
  * Manipulator 1.5.0
  * parts_library
  * QuickMeasure 2022.10.28
`

Neubo, 2023-10-20:
Hello,

if this commit and the new release 17 hours ago should solve the "Wrong CCX lib" issue, it doesn't.
Or do I test it in the wrong way ?

In the FEM/CalculX, If I point directly to the ccx.exe in freecad/bin or if I check "search in known binay directories", it failed.
If I point to the ccx.exe file in freecad/bin of the official/main freecad branch, it works.

Don't hesitate if it's not clear of if you need more details.

realthunder, 2023-10-21:
I have repackage the bundle. Please download again and try.

Neubo, 2023-10-22:
Thanks, I confirm, it works now !

closed 2023-10-30 by Neubo