realthunder/FreeCAD #864: AppImage Tip 2023.08.06: no solver backend found (usual fixes don't work) + request to retain old tip versions

eblanshey, 2023-08-09:
Hello,

I am having the common issue of no solver found, but all the usual fixes aren't working. I had the same issue on Windows, and I even [posted my fix here](https://github.com/realthunder/FreeCAD/issues/860#issuecomment-1662060480) about uninstalling and installing py_slvs.  But now I switched to Linux with the AppImage, and the solution doesn't work. I've tried everything: deleting all local FreeCAD config and cache directories, resetting preferences, installing py_slvs as my regular user, and as root, and upgrading the system's Python 3.10 to 3.11. The "Install Assembly Solver" popup keeps showing, and it fails to install.

```
21:23:00  pip install --user --disable-pip-version-check py-slvs
21:23:00  FreeCAD 0.21.0, Libs: 2023.806.0R14555 (Git shallow)
21:23:00  [FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
21:23:00  user site: /home/anon/.local/lib/python3.11/site-packages
21:23:00  Collecting py-slvs
21:23:00    Using cached py_slvs-1.0.6-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (316 kB)
21:23:00  Installing collected packages: py-slvs
21:23:00  Successfully installed py-slvs-1.0.6
21:23:00  1
21:23:00  
21:23:00  >>> >>> >>> >>> >>> >>> ... ... ... ... ... ... 
21:23:00  WARNING: Package(s) not found: py-slvs
21:23:00  
```

When I open the assembly, in the properties, the solver is set to None with no other options.
___
Everything works with the 2023.01.31 AppImage version out of the box, so a breakage was introduced recently. I would try to run more recent tip versions, like 2023.07.30, but because the "Tip" release is always overwritten, I don't have access to those assets. It would be great if we went back to creating a new release for each tip in order to save the asset history. If anyone can share the  07.30 AppImage with me, that would be great (email in profile).

I am running an Ubuntu 22.04-based Linux (Pop_OS).

```
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 2023.806.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 93dae74dba4a80dba52f9c5fe73fe5a874bae809
Python 3.11.4, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United States (en_US)
```

bambukouk, 2023-08-09:
Just noticed it...
I have also posted on the same? topic:
https://github.com/realthunder/FreeCAD_assembly3/issues/1314#issue-1842545203

shandyba, 2023-08-09:
Adding my vote for retaining the history of "tip" releases. The latest that worked for me flawlessly was the one released on May 21st, but I couldn't find assets for it anymore.
I'd love to try out new Tip releases, but with an option to be able to rollback, shall the Tip be too unsatisfactory.

eblanshey, 2023-08-09:
@bambukouk I'm aware they're all not "stable" but perhaps one is more stable than another :smile: indeed, being up to date with master would make it less stable.

I tried the edge version and it does work! Thanks for the tip.

QuakeString, 2023-08-10:
I've very same issue. 
```
08:10:27  pip install --user --disable-pip-version-check py-slvs
08:10:27  FreeCAD 0.21.0, Libs: 2023.806.0R14555 (Git shallow)
08:10:27  [FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
08:10:27  user site: /home/mozammel/.local/lib/python3.11/site-packages
08:10:27  Collecting py-slvs
08:10:27    Using cached py_slvs-1.0.6-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (316 kB)
08:10:27  Installing collected packages: py-slvs
08:10:27  Successfully installed py-slvs-1.0.6
08:10:27  1
08:10:27  
08:10:27  >>> >>> >>> >>> >>> >>> ... ... ... ... ... ... 
08:10:27  WARNING: Package(s) not found: py-slvs
08:10:27
```

I'm using Appimage
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2023.806.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 93dae74dba4a80dba52f9c5fe73fe5a874bae809
Python 3.11.4, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * BIM_Workbench 2021.12.0
  * A2plus 0.4.60n
  * dodo 1.0.0
  * FreeCAD_FastenersWB 0.4.59
  * FreeCAD_assembly3 0.12.0
  * PieMenu 1.2.7
[/code]

eblanshey, 2023-08-11:
Thank you @bambukouk, it works for me using Snap! I won't bother with the AppImage anymore for now.

bambukouk, 2023-08-14:
Just tested .11 appimage.
All seems fine to me :+1: 

[code]
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2023.811.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: d31fdf6ab71bf52a96d5d3b81f3b8349bca24e27
Python 3.11.4, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * ToolViewReverse_ToolBar
  * DynamicData 2.46.0
  * sheetmetal 0.2.49
  * Assembly3 0.11.4
  * fasteners 0.3.40
  * ThreadProfile 1.81.0
  * Manipulator 1.4.3
[/code]

thank you @realthunder  :1st_place_medal:

eblanshey, 2023-08-19:
Yes, it does work now out of the box, without even any prompts to install. Thank you!

closed 2023-08-19 by eblanshey