realthunder/FreeCAD #865: Appimage Tip 20230806 does not run under Wayland but does under X11

Tamagotono, 2023-08-09:
I am using Fedora 37 with Wayland and  tried the latest tip (20230806) which fails with a SEGFAULT.  If I switch to X11, it launches properly.
Using FreeCAD-Link-Tip-Linux-x86_64-py3.11-20230730.appimage and it works fine under both Wayland and X11, so the bug was just recently created.

`[jason@jason-fedora Applications]$ ./FreeCAD-Link-Tip-Linux-x86_64-py3.11-20230806_7d635b2a648168ae1b0c5f67c876bf8a.appimage 
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
/tmp/.mount_FreeCAXOk8Eq
FreeCAD 0.21.0, Libs: 2023.806.0R14555 (Git shallow)
xkbcommon: ERROR: failed to add default include path ././././././././././././././././././././././././../share/X11/xkb
failed to create xkb context
Failed to read /home/jason/.local/share/FreeCAD/Mod/ExtremeProLight/package.xmlQOpenGLFunctions created with non-current context
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3cb20) [0x7f879de7cb20]
#1  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_context_ref+0) [0x7f87966de960]
#2  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libxkbcommon.so.0(+0x1de05) [0x7f87966e0e05]
#3  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_keymap_new_from_buffer+0x47) [0x7f87966e05c7]
#4  0x7f8793cd8508 in QtWaylandClient::QWaylandInputDevice::Keyboard::keyboard_keymap(unsigned int, int, unsigned int) from /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0xe4
#5  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/./libffi.so.8(+0x6a4a) [0x7f879c652a4a]
#6  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/./libffi.so.8(+0x5fea) [0x7f879c651fea]
#7  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0xa0f8) [0x7f879c6620f8]
#8  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0x6949) [0x7f879c65e949]
#9  /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libwayland-client.so.0(wl_display_dispatch_queue_pending+0x74) [0x7f879c660124]
#10  0x7f8793cdc99b in QtWaylandClient::QWaylandDisplay::flushRequests() from /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x2d
#11  0x7f8793ce25ab in QtWaylandClient::QWaylandWindow::setVisible(bool) from /tmp/.mount_FreeCAXOk8Eq/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x127
#12  0x7f879ed0f878 in QWindowPrivate::setVisible(bool) from /tmp/.mount_FreeCAXOk8Eq/usr/bin/../lib/libQt5Gui.so.5+0x1ae
#13  0x7f879f38c2a0 in QWidgetPrivate::show_sys() from /tmp/.mount_FreeCAXOk8Eq/usr/bin/../lib/libQt5Widgets.so.5+0x2a8
#14  0x7f879f3920e6 in QWidgetPrivate::show_helper() from /tmp/.mount_FreeCAXOk8Eq/usr/bin/../lib/libQt5Widgets.so.5+0x12a
#15  0x7f879f393f61 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_FreeCAXOk8Eq/usr/bin/../lib/libQt5Widgets.so.5+0x203
#16  0x7f87a123d17d in Gui::MainWindow::startSplasher() from /tmp/.mount_FreeCAXOk8Eq/usr/bin/../lib/libFreeCADGui.so+0x26d
#17  0x7f87a0b6ffbb in Gui::postMainWindowSetup(Gui::MainWindow&) from /tmp/.mount_FreeCAXOk8Eq/usr/bin/../lib/libFreeCADGui.so+0x15ab
#18  0x7f87a0b7045b in Gui::Application::runApplication() from /tmp/.mount_FreeCAXOk8Eq/usr/bin/../lib/libFreeCADGui.so+0x13b
#19  /tmp/.mount_FreeCAXOk8Eq/usr/bin/FreeCADLink(+0x87e6) [0x564ff6a537e6]
#20  /lib64/libc.so.6(+0x27510) [0x7f879de67510]
#21  /lib64/libc.so.6(__libc_start_main+0x89) [0x7f879de675c9]
#22  /tmp/.mount_FreeCAXOk8Eq/usr/bin/FreeCADLink(+0x8bad) [0x564ff6a53bad]
`

Tamagotono, 2023-08-10:
I tested again with 20230810 and get the same error.

> `/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
/tmp/.mount_FreeCA9Diain
FreeCAD 0.21.0, Libs: 2023.810.0R14555 (Git shallow)
xkbcommon: ERROR: failed to add default include path ././././././././././././././././././././././././../share/X11/xkb
failed to create xkb context
Failed to read /home/jason/.local/share/FreeCAD/Mod/ExtremeProLight/package.xmlQOpenGLFunctions created with non-current context
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3cb20) [0x7f3eec27cb20]
#1  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_context_ref+0) [0x7f3ee2043960]
#2  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libxkbcommon.so.0(+0x1de05) [0x7f3ee2045e05]
#3  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libxkbcommon.so.0(xkb_keymap_new_from_buffer+0x47) [0x7f3ee20455c7]
#4  0x7f3ee20d8508 in QtWaylandClient::QWaylandInputDevice::Keyboard::keyboard_keymap(unsigned int, int, unsigned int) from /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0xe4
#5  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/./libffi.so.8(+0x6a4a) [0x7f3eeaa31a4a]
#6  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/./libffi.so.8(+0x5fea) [0x7f3eeaa30fea]
#7  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0xa0f8) [0x7f3eeaa410f8]
#8  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libwayland-client.so.0(+0x6949) [0x7f3eeaa3d949]
#9  /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libwayland-client.so.0(wl_display_dispatch_queue_pending+0x74) [0x7f3eeaa3f124]
#10  0x7f3ee20dc99b in QtWaylandClient::QWaylandDisplay::flushRequests() from /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x2d
#11  0x7f3ee20e25ab in QtWaylandClient::QWaylandWindow::setVisible(bool) from /tmp/.mount_FreeCA9Diain/usr/plugins/platforms/../../lib/libQt5WaylandClient.so.5+0x127
#12  0x7f3eed10f878 in QWindowPrivate::setVisible(bool) from /tmp/.mount_FreeCA9Diain/usr/bin/../lib/libQt5Gui.so.5+0x1ae
#13  0x7f3eed78c2a0 in QWidgetPrivate::show_sys() from /tmp/.mount_FreeCA9Diain/usr/bin/../lib/libQt5Widgets.so.5+0x2a8
#14  0x7f3eed7920e6 in QWidgetPrivate::show_helper() from /tmp/.mount_FreeCA9Diain/usr/bin/../lib/libQt5Widgets.so.5+0x12a
#15  0x7f3eed793f61 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_FreeCA9Diain/usr/bin/../lib/libQt5Widgets.so.5+0x203
#16  0x7f3eef63db0d in Gui::MainWindow::startSplasher() from /tmp/.mount_FreeCA9Diain/usr/bin/../lib/libFreeCADGui.so+0x26d
#17  0x7f3eeef6fe4b in Gui::postMainWindowSetup(Gui::MainWindow&) from /tmp/.mount_FreeCA9Diain/usr/bin/../lib/libFreeCADGui.so+0x15ab
#18  0x7f3eeef702eb in Gui::Application::runApplication() from /tmp/.mount_FreeCA9Diain/usr/bin/../lib/libFreeCADGui.so+0x13b
#19  /tmp/.mount_FreeCA9Diain/usr/bin/FreeCADLink(+0x87e6) [0x557d8585d7e6]
#20  /lib64/libc.so.6(+0x27510) [0x7f3eec267510]
#21  /lib64/libc.so.6(__libc_start_main+0x89) [0x7f3eec2675c9]
#22  /tmp/.mount_FreeCA9Diain/usr/bin/FreeCADLink(+0x8bad) [0x557d8585dbad]
`

realthunder, 2023-08-11:
Can you please try extract the AppImage using the following command

```
./FreeCAD-Link-Tip-Linux-x86_64-py3.11-20230810.AppImage --appimage-extract
```

This will extract the AppImage content into a directory called `squashfs-root`. Replace the file `squashfs-root/AppRun` with [this](https://github.com/realthunder/FreeCADMakeImage/raw/master/conda/AppDir/AppRun). And then try to run it with

```
cd squashfs-root
./AppRun
```

Tamagotono, 2023-08-14:
@realthunder Thank you for your reply. I did not see it until after testing the 20230811 version, which works fine.
Would it be of help to you if I did as you suggest with the 20230810 version, at this point?  If not, then I will close this issue.

On a side note, thank you for all of your work that you do on this project.

realthunder, 2023-08-14:
You're welcome. No need to test, as I've included the mentioned modification in the latest release.

closed 2023-08-14 by realthunder