realthunder/FreeCAD #869: Unpack problem for the windows version

dingolino, 2023-08-12:
Hi, since a couple of Versions I am getting a error during unpacking the windows version, that files are missing.

Please see attached pictures. I can sjip it and umpacking continou, but it oes not look "nice" getting a error meassage during unpcking.

[image]
[image]
[image]
[image]

Thorn72, 2023-08-15:
hello,

I don't have this problem, I'm on windows 10

dingolino, 2023-08-18:
Hi,
was digging a little bit deeper. Please see attached screen shot.
I updated my 7zip from 22.01 to 23.01 (it was not "that old") and now the error meassage is gone.

Best regards,

[image]

closed 2023-08-18 by dingolino