realthunder/FreeCAD #870: FreeCAD Link Tip 2023.08.11 - PAD/Direction axis/Custom direction is not working

6rzegorz, 2023-08-13:
Hello I'm FreeCAD Link user, designing carbodies is my hobby. 
Last few Tip versions has problem with setting a custom direction of Pad operation (x, y, z. fields are blocked). 
Could you correct it? Thank you in advance.

vovodroid, 2023-09-25:
Confirming:

[image]

The same strange thing happens on returning to Normal:

[image]

Working workaround: click OK and reopen PAD dialog.

6rzegorz, 2023-09-26:
>  Confirming:
>  user-images.githubusercontent.com

Hello.  Thank you for your answer and your help. It seams that this bypass is really working. But in the next FreeCAD Link version it would be nice to see it corrected.   

 user-images.githubusercontent.com