realthunder/FreeCAD #876: Wrong behavior of the Pocket feature

AlexStOd, 2023-08-22:
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2023.811.0.38816 (Git) Snap 117
Build type: Release
Branch: tag: 20230811tip
Hash: d31fdf6ab71bf52a96d5d3b81f3b8349bca24e27
Python 3.8.10, Qt 5.15.7, Coin 4.0.1, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

I am facing with a strange issue - when I try to create a pocket (subtractive) - it creates pad (additive) instead.
Steps to reproduce:
- create some sketch on one of the existing body faces
[image]
- make a pocket
[image]

Expected behavior:
Hole is created in the mentioned face:
[image]

Actual behavior:
Round body is shown instead:
[image]

[DrumSensorHolder.zip](https://github.com/realthunder/FreeCAD/files/12406035/DrumSensorHolder.zip)

AlexStOd, 2023-08-22:
May it be related on my OS/installed software? For example - wrong solver lib, etc.
The similar issue is periodically appearing in different projects. But it cannot be reproduced on simple examples.

AlexStOd, 2023-08-22:
Actually, there is a workaround - change operations order. For example, it works fine without fillets before it.
But there is still a lot of cases where changing of operations order is sensitive. So it would be great to fix this bug.

realthunder, 2023-08-27:
This is most likely OCC the CAD kernel problem. The problem starts with `Fillet001`. You can select the object and click geometry check to see the error. I am not sure why OCC has trouble with this seemingly simple fillet operation. BTW, what is `Fillet` here for?

[image]

Make sure to select `Run boolean operation check`.

[image]

AlexStOd, 2023-08-27:
Hm.. I tried to do CheckGeometry with the mentioned settings. It showed me a number of errors:
[image]

But how can I fix / avoid them? 

>BTW, what is Fillet here for?
It's here to smooth internal edges of the "roof".