realthunder/FreeCAD #878: libicuuc.so.72: cannot open shared object file: No such file or directory

chiboreache, 2023-08-25:
```sh
FreeCAD 0.20, Libs: 0.20R31683 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
#####                 ####  ###   ####
#                    #      # #   #   #
#     ##  #### ####  #     #   #  #   #
####  # # #  # #  #  #     #####  #   #
#     #   #### ####  #    #     # #   #
#     #   #    #     #    #     # #   #  ##  ##  ##
#     #   #### ####   ### #     # ####   ##  ##  ##

libicuuc.so.72: cannot open shared object file: No such file or directory
```

```sh

➜  ~ pacman -Qo /usr/lib/libicuuc.so
/usr/lib/libicuuc.so is owned by icu 73.2-2
➜  ~ pac icu
warning: icu-73.2-2 is up to date -- reinstalling
```

```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.31683 (Git)
Build type: Release
Branch: makepkg
Hash: cad29253a493084a1c52d09fe164d4d91428b265
Python version: 3.11.3
Qt version: 5.15.10
Coin version: 4.0.1
OCC version: 7.6.3
Locale: English/United States (en_US)
```

realthunder, 2023-08-27:
I don't recognize the build information. Where did you download the package?

chiboreache, 2023-09-04:
@realthunder 

it's from [AUR](https://wiki.archlinux.org/title/Arch_User_Repository) 

> The Arch User Repository (AUR) is a community-driven repository for Arch users. It contains package descriptions ([PKGBUILDs](https://wiki.archlinux.org/title/PKGBUILD)) that allow you to compile a package from source with [makepkg](https://wiki.archlinux.org/title/Makepkg) and then install it via [pacman](https://wiki.archlinux.org/title/Pacman#Additional_commands).

here is the [package](https://aur.archlinux.org/packages/freecad-linkstage3-git) and it's sourcing from here

[image]