realthunder/FreeCAD #880: Can't install addon in latest 20230811

vovodroid, 2023-08-30:
[image]

```
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2023.811.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: d31fdf6ab71bf52a96d5d3b81f3b8349bca24e27
Python 3.11.4, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English
```

Marcin-Quosoo-Klos, 2023-08-30:
Hi,

Go to Edit/Preferences/Add-on Manager and check "disable git".

Cheers,
Marcin

vovodroid, 2023-08-30:
> check "disable git".

Yes, this helps. Anyway something changed from previous versions.

Miksu80, 2023-11-27:
Got the same with 2023.1102.0 build 14555 and windows 11, got over with "disable git"..