realthunder/FreeCAD #882: Current linkmerge build can't find asm3

wsteffe, 2023-09-01:


wsteffe, 2023-09-05:
To better clarify the issue:
 In my ubuntu 22.04 asm3 is installed in /home/<your_user_name>/.FreeCAD/Mod follwing RT instructions and it is properly loaded by Linkdaily. But Linkmerge (which I built from sources as I did with Linkdaily) can't find it.

wsteffe, 2023-09-12:
Now I have moved asm3 from "/home/<your_user_name>/.FreeCAD/Mod" to my FreeCAD installation directory "/usr/local/Mod" and Linkmerge can find it.

But the install instruction given at https://github.com/realthunder/FreeCAD_assembly3
are not valid for Linkmerge.

realthunder, 2023-09-12:
As inherited from change in upstream, `.FreeCAD` is no longer used as user data storage. Instead, please use `.local/share/FreeCAD`. You can either copy or symlink `.FreeCAD/Mod` directory in `.local/share/FreeCAD`.