realthunder/FreeCAD #884: Shadow Draw Style seems broken on MacOS Ventura (Apple Silicon)

ademlabs, 2023-09-03:
Using version 2023.816.0, revision 14555 on MacOS, the shadow draw style renders in weird colours. When rotating the view the colours change to various shades of colours. This seems to be present in older versions as well. 

[image]
[image]
[image]

Solar-Mann, 2024-08-06:
Same here with the last 3 versions of FreeCAD ASM3. Hardy useful with such a glitch...

Solar-Mann, 2024-10-21:
Beside Assembly 3 the shadow rendering capabilities have been the main reason to use Linkstage. This has become useless. How can such a flaw not be fixed over several months...