realthunder/FreeCAD #886: Frequesnt segmentation faiult in Tip 2023.08.11

ofpkubi, 2023-09-09:
Hi,

I have quite frequent segfaults in 20230811 Tip version. I can't see any correlation with activities in FreeCAD or any files I load. It is quite random.

No crashes with 20230521 Tip.

System:
Debian testing (trixie), AMD64
FreeCAD-Link-Tip-Linux-x86_64-py3.11-20230811.AppImage 
python3        3.11.4-5+b1
python3-pyqt5             5.15.9+dfsg-2
python3-pyqt6              6.5.2-1 
libqt5*          5.15.10
libqt6*          6.4.2

---------

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c510) [0x7fb27f877510]
#1  0x7fb282993e63 in SoFCVertexArrayIndexer::getBoundingBox(SbMatrix const*, SbBox3f&, SbVec3f const*) const from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x23
#2  0x7fb2829ccbe4 in SoFCRenderCache::buildHighlightCache(std::map<int, Gui::CoinPtr<SoFCVertexCache>, std::less<int>, std::allocator<std::pair<int const, Gui::CoinPtr<SoFCVertexCache> > > >&, int, SoDetail const*, unsigned int, int) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x1d94
#3  0x7fb2829dadc8 in SoFCRenderCacheManagerP::updateSelection(void*, SoSensor*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x2b8
#4  0x7fb2829dbded in SoFCRenderCacheManager::addSelection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, SoPath*, SoPath*, SoDetail const*, unsigned int, bool, bool, bool) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0xacd
#5  0x7fb282a132e6 in Gui::SoFCUnifiedSelection::Private::checkSelection(Gui::SelectionChanges::MsgType, App::SubObjectT const&) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x566
#6  0x7fb282a2981e in Gui::SoFCUnifiedSelection::Private::doAction(SoAction*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x34e
#7  0x7fb282a29ec1 in Gui::SoFCUnifiedSelection::doAction(SoAction*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x51
#8  0x7fb27f0c7118 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/./libCoin.so.80+0x68
#9  0x7fb27f0c7dfa in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/./libCoin.so.80+0x2ea
#10  0x7fb282b1c384 in Gui::View3DInventorViewer::onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x854
#11  0x7fb282d40410 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x30
#12  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so(+0x90088d) [0x7fb28290088d]
#13  0x7fb282d41985 in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x465
#14  0x7fb282d420d7 in Gui::SelectionSingleton::flushNotifications() from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x1a7
#15  0x7fb282d4220b in Gui::SelectionPauseNotification::~SelectionPauseNotification() from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x2b
#16  0x7fb2829202c9 in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x169
#17  0x7fb282925c4e in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x19e
#18  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5(+0x1dff9b) [0x7fb2801dff9b]
#19  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5(+0x33836b) [0x7fb280f3836b]
#20  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5(+0x1dfff5) [0x7fb2801dfff5]
#21  0x7fb28019ac3a in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0x2a
#22  0x7fb28019fd71 in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0x1c3
#23  0x7fb280f31786 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0x4a4
#24  0x7fb280f319d3 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0x201
#25  0x7fb280eef60c in QAbstractItemView::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0x2ec
#26  0x7fb28292ff6c in Gui::TreeWidget::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x14c
#27  0x7fb280d941ee in QWidget::event(QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0x120
#28  0x7fb280dfe7d7 in QFrame::event(QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0x27
#29  0x7fb2801c2880 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0xe6
#30  0x7fb280d6a076 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0xc8
#31  0x7fb280d6e36d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0x663
#32  0x7fb28260ef99 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x99
#33  0x7fb2801c2a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0x8e
#34  0x7fb280d6d981 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0x283
#35  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5(+0x1a338d) [0x7fb280da338d]
#36  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5(+0x1a4505) [0x7fb280da4505]
#37  0x7fb280d6a087 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#38  0x7fb28260ef99 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x99
#39  0x7fb2801c2a18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0x8e
#40  0x7fb28070956e in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Gui.so.5+0x6b6
#41  0x7fb2806f6091 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Gui.so.5+0x57
#42  /tmp/.mount_FreeCAinyzp4/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x5672c) [0x7fb27522072c]
#43  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x2ad) [0x7fb27e51538d]
#44  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/./libglib-2.0.so.0(+0x5d649) [0x7fb27e515649]
#45  /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fb27e5156de]
#46  0x7fb2801f3d50 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0x5c
#47  0x7fb2801c05d8 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0x180
#48  0x7fb2801c47af in QCoreApplication::exec() from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libQt5Core.so.5+0x107
#49  0x7fb282570666 in Gui::Application::runApplication() from /tmp/.mount_FreeCAinyzp4/usr/bin/../lib/libFreeCADGui.so+0x4b6
#50  ./FreeCADLink(+0x87e6) [0x55688a89e7e6]
#51  /lib/x86_64-linux-gnu/libc.so.6(+0x276ca) [0x7fb27f8626ca]
#52  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7fb27f862785]
#53  ./FreeCADLink(+0x8bad) [0x55688a89ebad]

eblanshey, 2024-01-06:
I'm having the same issue, but with the latest stable 2024-01-01. It seems random, because after restarting the application and doing the same action that triggered it before, there is no crash. I did not have this issue with 2023.11.02. Running using AppImage on Ubuntu.

In this particular crash I was in a sketch and deleted an imported line from the 3d geometry.

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f7537e37520]
#1  0x7f75377e30a5 in SoField::valueChanged(int) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/./libCoin.so.80+0x45
#2  0x7f74091e3ac0 in SketcherGui::ViewProviderSketch::updateColor() from /tmp/.mount_FreeCA1a2l3I/usr/lib/SketcherGui.so+0x11a0
#3  0x7f74091ea624 in SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCA1a2l3I/usr/lib/SketcherGui.so+0xa34
#4  0x7f753b346c20 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADGui.so+0x30
#5  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADGui.so(+0x90479d) [0x7f753af0479d]
#6  0x7f753b346e8a in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADGui.so+0x22a
#7  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADGui.so(+0x90479d) [0x7f753af0479d]
#8  0x7f753b3481e5 in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADGui.so+0x465
#9  0x7f753b35306f in Gui::SelectionSingleton::addSelection(char const*, char const*, char const*, float, float, float, std::vector<Gui::SelectionSingleton::SelObj, std::allocator<Gui::SelectionSingleton::SelObj> > const*, bool) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADGui.so+0x9cf
#10  0x7f74091c6f99 in SketcherGui::ViewProviderSketch::selectElement(char const*, bool) const from /tmp/.mount_FreeCA1a2l3I/usr/lib/SketcherGui.so+0x599
#11  0x7f74091964a2 in SketcherGui::TaskSketcherElements::on_elementsWidget_itemSelectionChanged() from /tmp/.mount_FreeCA1a2l3I/usr/lib/SketcherGui.so+0x9d2
#12  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5(+0x1dfff5) [0x7f75387dfff5]
#13  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Widgets.so.5(+0x33836b) [0x7f753953836b]
#14  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5(+0x1dfff5) [0x7f75387dfff5]
#15  0x7f753879ac3a in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5+0x2a
#16  0x7f753879fd71 in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5+0x1c3
#17  0x7f753879be23 in QItemSelectionModel::select(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5+0x2f
#18  0x7f753879ce3e in QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5+0xfa
#19  0x7f75394ec1c3 in QAbstractItemView::setCurrentIndex(QModelIndex const&) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Widgets.so.5+0x5d
#20  0x7f740919b8f4 in SketcherGui::TaskSketcherElements::slotElementsChanged() from /tmp/.mount_FreeCA1a2l3I/usr/lib/SketcherGui.so+0x3d4
#21  /tmp/.mount_FreeCA1a2l3I/usr/lib/Sketcher.so(+0xec386) [0x7f740bcec386]
#22  0x7f740bd130f6 in Sketcher::SketchObject::onChanged(App::Property const*) from /tmp/.mount_FreeCA1a2l3I/usr/lib/Sketcher.so+0x3a6
#23  0x7f7539c65238 in App::Property::touch() from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADApp.so+0x1a8
#24  0x7f740bd0dc49 in Sketcher::SketchObject::delGeometriesExclusiveList(std::vector<int, std::allocator<int> > const&) from /tmp/.mount_FreeCA1a2l3I/usr/lib/Sketcher.so+0x559
#25  0x7f740bd0e258 in Sketcher::SketchObject::delGeometries(std::vector<int, std::allocator<int> > const&) from /tmp/.mount_FreeCA1a2l3I/usr/lib/Sketcher.so+0x208
#26  0x7f740bdcd57f in Sketcher::SketchObjectPy::delGeometries(_object*) from /tmp/.mount_FreeCA1a2l3I/usr/lib/Sketcher.so+0x2af
#27  0x7f740bdcdba6 in Sketcher::SketchObjectPy::staticCallback_delGeometries(_object*, _object*) from /tmp/.mount_FreeCA1a2l3I/usr/lib/Sketcher.so+0x56
#28  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(+0x2020d8) [0x7f753a2020d8]
#29  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x25c) [0x7f753a1e2cec]
#30  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f753a1ed3d2]
#31  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(+0x2a6e3d) [0x7f753a2a6e3d]
#32  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f753a2a68b4]
#33  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(+0x2c5ae3) [0x7f753a2c5ae3]
#34  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(+0x2c1c0a) [0x7f753a2c1c0a]
#35  /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f753a2b5b09]
#36  0x7f753833c01c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADBase.so+0x6c
[... cut ...]
#62  0x7f75387c05d8 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5+0x180
#63  0x7f75387c47af in QCoreApplication::exec() from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libQt5Core.so.5+0x107
#64  0x7f753ab72c26 in Gui::Application::runApplication() from /tmp/.mount_FreeCA1a2l3I/usr/bin/../lib/libFreeCADGui.so+0x4b6
#65  ./FreeCADLink(+0x87e6) [0x560879c337e6]
#66  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f7537e1ed90]
#67  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f7537e1ee40]
#68  ./FreeCADLink(+0x8bad) [0x560879c33bad]

realthunder, 2024-01-07:
This is not the same crash as the OP. For yours, it is fixed in the new release 2024.01.04

eblanshey, 2024-01-09:
I didn't notice the new release, thank you!