realthunder/FreeCAD #891: [FR] Angle constrain between curves

vovodroid, 2023-09-20:
It would be useful having option to measure angle between curves:

[image]

Actually it should be angle between tangential lines.

bombela, 2023-09-23:
You can use a tangent construction line for this.

vovodroid, 2023-09-23:
That's what I do know. But direct constraints will be better.

Trakyan, 2023-10-08:
An angle to a curve doesn't really make sense. You'd first need to specify a point on the circle, and whether you want to measure from the normal or tangent at that point. Both require construction geometry anyway.

vovodroid, 2023-10-09:
> You'd first need to specify a point on the circle

Sure. My example implies tangent. I guess it should be default mode.