realthunder/FreeCAD #892: Forced, inexplicable PartDesign migration breaks the object tree

m-sundman, 2023-09-30:
I've created a project completely in the 2023.811.0 (r14555), and suddenly trying to create a new sketch in one of the 4 PartDesign bodies it shows a message saying:
> The document "Test" seems to be either in the middle of the migration process from legacy PartDesign or have a slightly broken structure. Do you want to make the migration automatically?

And under "Show Details..." it says:

> Note: If you choose to migrate you won't be able to edit the file with an older FreeCAD version.
> If you refuse to migrate you won't be able to use new PartDesign features like Bodies and Parts. As a result you also won't be able to use your parts in the assembly workbench.
> Although you will be able to migrate any moment later with 'Part Design -> Migrate'.

It shouldn't be in the middle of some migration from any legacy thing, because it only ever was used on this same version of FreeCAD, but it seems I can't make more Sketches without migrating.

If I migrate (either by selecting "Yes" in the dialog, or by menu Part Design -> Migrate), it creates a new Part object containing copies of each PartDesign body, each one containing the same features as the original body:
[image]

If I delete the body copies (under Part) then all features will also be deleted from the original bodies.

Also, can I repair this project somehow?
For a copy of the project, send me a message at sundman (ät) iki (döt) fi or on patreon.

realthunder, 2023-10-15:
Could you please post your project, or send me by email in private.

m-sundman, 2023-10-15:
Sent per email, sorry for the big file. (Let me know if the message is too big and I should upload it to some file sharing site instead.)

m-sundman, 2023-10-22:
With the newest version it works if I first recompute each object before trying to make any sketches. (If I just try to make a sketch without first recomputing then it will say I need to migrate, and that results in the problems stated above, but if I recompute first then I won't have to migrate.)

closed 2023-10-22 by m-sundman