realthunder/FreeCAD #896: Bug With Editing "Value" Fields In Property Browser

Trakyan, 2023-10-19:
In the latest tip release 2023.10.18 on windows (will check and confirm linux when I get home). Both py 3.11 and 3.8 versions seem to exhibit this.

The bug occurs when trying to edit the "value" field in the property viewer. It highlights/selects the entire text field after each keystroke, so the next keystroke deletes/replaces the contents already there. 

A couple of work arounds are using the expression editor to enter values, or using the mouse to click the entry field and deselect/unhighlight the contents between each keystroke.

Can anyone else recreate this?

realthunder, 2023-10-27:
Can any of you provide a screencast showing the problem. Also, please post your configuration file here, `Tools -> Edit parameters -> Export`. Thanks!

GinTonic99, 2023-10-27:
Enclosed a screencast. I selected the lenght field and entered 1 2 3 4, expecting that at the end the field should contain 1234. Enclosed also the configuration export.
Hope this helps and thank you in advance

https://github.com/realthunder/FreeCAD/assets/34986229/b79fc7d5-1c15-440f-bc64-a3dda6966deb

[Freecad Parameter 2023 10 27.FCParam.zip](https://github.com/realthunder/FreeCAD/files/13189905/Freecad.Parameter.2023.10.27.FCParam.zip)

mzient, 2023-10-28:
Bumping up the issue! The only way to edit properties is to do so in an external editor and paste.

GinTonic99, 2023-10-30:
resolved in 
[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2023.1029.0.38882 (Git) Snap 120
Build type: Release
Branch: tag: 20231029tip
Hash: 471e6aca2118969933ddfcc71a03fca771de2464
Python 3.8.10, Qt 5.15.7, Coin 4.0.1, Vtk 7.1.1, OCC 7.6.3
Locale: German/Austria (de_AT)
Installed mods: 
  * sheetmetal 0.3.1
  * Curves 0.6.13
  * fasteners 0.4.67
  * lattice2 1.0.0
  * QuickMeasure 2022.10.28
  * Assembly4 0.50.4
  * FreeCAD_SketchArch
  * CurvedShapes 1.0.4
  * Render 2023.8.18
[/code]

many thanks for the quick repair

commit 4b628269 (2023-10-28): Gui: do not change property view editing content

Fixes #896
closed 2023-11-01 by Trakyan