realthunder/FreeCAD #898: Edge rendering broken

mgth, 2023-10-21:
Using Tip 2023.10.18 in windows 
FreeCAD-Link-Tip-Win-x86_64-py3.11-20231018.7z 

Edge rendering is ok when mouse is in viewport, but completely broken when the pointer leave the zone :

[image]

[image]

mgth, 2023-10-21:
After some testing only auto/distributed render cache is affected.
Centralized and experimental are ok.

realthunder, 2023-10-22:
Could you please post the file, or maybe reproduce the problem with a simplified one?

mgth, 2023-10-23:
yes, on simple cube the vertex joins wrong corners :
[cube.zip](https://github.com/realthunder/FreeCAD/files/13068395/cube.zip)
[image]

realthunder, 2023-10-24:
I can't reproduce the problem. Could you please send me your user config, `Tools -> Edit parameters -> Export`.

mgth, 2023-10-25:
[mgth.zip](https://github.com/realthunder/FreeCAD/files/13162436/mgth.zip)

alexawka, 2024-03-23:
I have encountered same problem.
When rotating object, edges get jumbled up. When hovering or clicking on object - render fixes itself until next rotation.
Changing Render Cache to Experimental fixed the issue.
FreeCad official does not seem to have same issue.

PC is Windows 11 and AMD GPU with latest drivers.
Prior on two Windows 10 and NVidia GPU it was not encountered.

Maybe windows version of gpu drivers are at fault?

realthunder, 2024-03-24:
Yes, I can reproduce this on an AMD Windows machine, which leads me to think it is an OpenGL driver compatibility problem. I have yet to find any fix for this. Meanwhile, please try to change `Preference -> Display -> Render caching` to `Experiemental`. It works fine on my machine.