realthunder/FreeCAD #899: Wire is not closed in PD Loft

m-sundman, 2023-10-21:
The Loft function of Part Design is confused. Each sketch is closed (and can e.g. be extruded successfully), but trying to make a loft of them fails with a "wire is not closed" error message.

[Loft.FCStd](https://github.com/realthunder/FreeCAD/files/13061367/Loft.FCStd.zip)

realthunder, 2023-10-22:
It is because you are selecting individual edges when creating the Loft. To select the whole shape, you can select in the tree view, or use `G, G` command to select higher level geometry in 3D view.

[image]

m-sundman, 2023-10-24:
> It is because you are selecting individual edges when creating the Loft. To select the whole shape, you can select in the tree view, or 

If I select sketches in the tree view FreeCAD crashes completely.

m-sundman, 2025-05-09:
Fixed.

closed 2025-05-09 by m-sundman