realthunder/FreeCAD #908: Crashing in Assembly 3 when using Groups to sort parts

stoky996, 2023-11-12:
To Whom it may concern

love using assembly 3 on this version of Freecad follows the constraint theme of the sketcher and part design very nicely. I am however running into an issue regarding using group folders to sort parts out in an assembly when dragging the parts into the group folder only on some specific parts despite them being identical copies can crash Freecad i say can cause the frequency can vary ALOT sometimes just changing the order that i import them in i am able to get by and i simply save every time i try to group parts but it would be a lot nicer if it was fixed i could put them in the group beforehand but most of the time i end up forgetting to do so or it crashes when i try to import the group into the parts section of the assembly. So i am wondering if this is a known issue or if this is something worth fixing and what to do about it? if you need a crash log then i will try to provide them just list how to as i am not sure how to find the crash log.

regards jesse

realthunder, 2023-12-28:
What type of group are you using? Assembly3 has its own type for grouping parts/elements/constraints, although using plain group shouldn't cause crash. 

[image]

Could you provide some sample file and screenshots? It will also help if you could provide the crash dump. If you are using linux, simply launch the application in a terminal. It will print out crash dump in the terminal. If you are using Windows, try search for `crash.dmp` file in your user directory, usually `C:\Users\<your_username>\AppData\Roaming\FreeCAD`.

stoky996, 2023-12-28:
oh ok think i found out why it was crashing will post example crash log anyway if it happens again i was using the grouping function that is included with the file tab (the one with the create part function) did not notice that there was a separate one for assembly 3 till i looked at the pic you provided.

I should mention since posting my previous post I since changed back to linux from windows for different reasons and i am now using the pre release flatpak version and it does not seem to crash for now at least in the two assemblies i have since tried but thanks for responding hope i did not waste much time. I will post again in here if it happens again but as for now i can't recreate it.

regards jesse.

closed 2023-12-28 by stoky996