realthunder/FreeCAD #909: First Keypress in "Dimension Distance" Tool Ignored

Trakyan, 2023-11-18:
This is a bug that's been around for a while. For now I can confirm it exists in the latest 02/11/2023 linux appimage and I've seen it recent windows releases (but currently can't confirm if it's in the latest windows release). It's existed at least since the current stable release from january 2023.

The bug is as follows: When I use the keyboard shortcut for "dimension distance" (D, D in my case), and "ask for value after creating dimensional constraint" is ticked, if I click a line FIRST and then press D, D the dialogue box opens up, and now the first keypress is ignored. Opening the dimension box to edit it a second time doesn't have this bug.

Just now I did some playing around to try and narrow down the bug, and discovered quite a few things that cause it not to trigger. 
-If the "ask for value after creating dimensional constraint" is not ticked, even the first time you open the dimension box to edit it, the first keypress will work.
-If using the dimension distance tool from the icon on the toolbar, the bug doesn't trigger no matter what I do.
-If I press, D,D to trigger the dimension distance tool, and then click the line to be dimensioned, the bug does not trigger and the first keypress registers fine.

Changing the shortcut from D,D to D doesn't seem to affect things

As a side note, the addon manager and rick clicking the toolbar->customize have been causing my appimages to crash for a while. I've been getting around this by using the mainline freecad appimage to update addons and to modify the toolbar. Is there some kind of conflict between the two version? This has been the case for me since the january release (january was the last "working" one, though it had a different bug where custom toolbars would randomly be deactivated or deleted).

realthunder, 2023-12-28:
Can you please try the [Tip](https://github.com/realthunder/FreeCAD/releases/tag/Tip) release. I can't seem to reproduce the problem. Also, can you please check what is your settings in `Tools -> Customize -> Keyboard -> Multi-key sequence delay`? This setting affects the delay for recognizing multi-key sequence shortcut with ambiguation.

Trakyan, 2023-12-29:
Like I said in the OP it is present in 11.02 tip version. I haven't been able to open the customize menu for quite a few versions (I think back since the jan 2023 release?) since it crashes the app image, same with the addon manager. As far as I can tell the delay is set to 300ms. The bug is present as well when I set the shortcut to just a single D not D,D, or when using another letter as well (I tried Q which I'm not using for anything else).

To be clear, the bug _only_ occurs when the following are done in order: click a line, press the keyboard shortcut (D), then the first keypress is ignored. 

Using the shortcut first, then clicking the line, or using the toolbar to trigger the tool cause it to behave normally. I've had a few people mention they have the same bug, but with the other dimension tools, it's only been with the dimension distance tool for me though. Changing the multi-key sequence delay doesn't seem to have any effect, and if I wait (i.e. 5+ seconds, clearly long enough enough to pass the 300 ms delay) before the first keypress it also makes no difference, it's still ignored. I also tried with the delay set to 1, and still no luck.

Trakyan, 2023-12-30:
> Can you please try the [Tip](https://github.com/realthunder/FreeCAD/releases/tag/Tip) release. I can't seem to reproduce the problem. Also, can you please check what is your settings in `Tools -> [...]

I've done a bit more testing, and the bug is even more situational than I thought. I noticed the bug doesn't trigger with the default K,D shortcut. So I've gone through and tested all the letters to see which ones trigger the bug. At first I thought only the first letter affect the bug, since the default shortcut K,D does not trigger the bug. D,D and D both trigger the bug, but K,D does not despite having a D in it. However K triggers the bug, while K,D does not.

The following single key shortcuts  (I tested letters only) DO trigger the bug (select line, then press shortcut key and first keypress will be ignored): Q, W, R, T, U, S, D, G, K, Z, X, C, V, M.
Other single key shortcuts  work fine, without triggering the bug. 

Using these shortcuts with the other constrain/dimension tools (i.e. diameter) also seems to trigger the same bug of the first keypress being ignored. I only tested K and "constrain diameter" though so I can't be sure if the exact same shortcuts affect each of the dimension tools.

commit 9119c9c5 (2023-12-30): Gui: fix lost key when resolving multi-key-sequence shortcut

Fixes #909
commit bd5f8258 (2023-12-31): Gui: fix lost key when resolving multi-key-sequence shortcut

Fixes #909
closed 2023-12-31 by Trakyan