realthunder/FreeCAD #913: Objects are shown where they shouldnt

ElPatoNoEsTu, 2023-11-30:
Hi, 

there might be an "error"  ?!  - Imho - the shapbinder in Body001 seems to be visible while it shouldnt.

[image]
[visibility.zip](https://github.com/realthunder/FreeCAD/files/13518649/visibility.zip)

Version used:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 2023.1102.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 53783b6707f2fd2d561a02b5542f4fbe3c1eda4a
Python 3.11.6, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus.backup1698606179.0777142
  * Assembly3.backup1701122976.354356 0.11.3
  * Curves.backup1701123223.0435972
  * LCInterlocking
[/code]

Regards, ElPatoNoEsTu

realthunder, 2023-12-25:
Because you are using the ShapeBinder as the base feature for `Pad001`. If that's not what you want, you can double click the pad, and change `Operation` from `Additive` to `New shape`.

ElPatoNoEsTu, 2023-12-26:
Thanks a lot - This was the missing link !!!  
I must admit, that I expected "New Shape" as default behavior, using the shape-binder just to get some geometry from the first body.

closed 2023-12-26 by ElPatoNoEsTu