realthunder/FreeCAD #916: Bug with Wireframe Rendering

samin50, 2023-12-21:
Hi,
I have encountered a weird bug, and I think it happened right after I uninstalled the official version of FreeCAD to free up space.
For some reason I get these random lines coming out of my geometry, and I believe it is a wireframe issue; if I change the draw style to wireframe only I see these lines, but I do not if I switch it to shaded.
[image]
[image]

I am not actually sure if the uninstall of the official version of FreeCAD caused this, but thats the only thing I could think of. I reinstalled it and downloaded the latest version of realthunder but no change.

realthunder, 2023-12-22:
Could you please attach the file here? Also please post your setting file, `Tools -> Edit parameters -> Export`.

Please also try `Tools -> Preset Configurations -> Display settings` and see if it helps.

samin50, 2023-12-22:
That seemed to work wow, can't believe it was that simple. I should just mention that it happened across all files too not just my local one. I would still attach my setting file but I attempted your fix first and I feel whatever was wrong with my display file has now been fixed by your suggestion

closed 2023-12-22 by samin50