realthunder/FreeCAD #918: No Windows build in latest release

vovodroid, 2024-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Latest
```

### Subproject(s) affected?

None

### Problem description

No Windows asset here: https://github.com/realthunder/FreeCAD/releases/tag/20231231stable

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jj-uk, 2024-01-02:
The windows builds are now there but 7-zip reports an error when extracting

https://github.com/realthunder/FreeCAD/releases/download/20231231stable/FreeCAD-Link-Stable-Win-x86_64-py3.11-20231231.7z

vovodroid, 2024-01-02:
I opened and run it, try download again.

vovodroid, 2024-01-02:
But I used WinRar, not 7zip.

jj-uk, 2024-01-02:
7zip was out of date. Updated and now all ok.

closed 2024-01-03 by vovodroid