realthunder/FreeCAD #920: [Misc] Link to Windows LibPack used in 20231231stable build?

scottmudge, 2024-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Related to build 20231231stable
```

### Subproject(s) affected?

None

### Problem description

The LibPack I'm using to build this fork of FreeCAD is quite old. Not wanting to recompile everything again, I was wondering if you would be willing to supply a link to the LibPack used for your Windows builds at `20231231stable` with Python311. 

Comparing the release to the "official" FreeCAD libpack, the official libpack looks a bit out of date. E.g., VTK is older than the 9.2 used in `20231231stable`, boost is older, etc. So to keep things consistent with the builds supplied here, it would be nice to build against the same dependencies/libraries.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-01-02:
The Windows release is built using conda.

scottmudge, 2024-01-02:
Cool, thanks!

closed 2024-01-02 by scottmudge

arthurx70, 2024-04-02:
so, where is the LibPack?

scottmudge, 2024-04-02:
Looks like the Windows builds here are just using an Anaconda environment as part of the CI/CD pipeline. I couldn't get it working by manually setting up a conda environment, so I rebuilt/compiled all the dependencies manually and created my own libpack.

arthurx70, 2024-04-02:
wow! cool!!
but realthunder  released  windows version LinkStable, he has compiled all the dependencies manually I guess.
Would you like to share your LibPack?
thx!