realthunder/FreeCAD #922: [Problem] Lots of crashes in Sketcher

m-sundman, 2024-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2024.101.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 97ccc0b4c01d671f5a85116ffa5f71464f1eee1c
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.8
  * CurvedShapes 1.0.4
  * ThreadProfile
  * fasteners 0.4.54
  * fcgear
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Frequent crashes for very common operations in sketcher (at least inside a Part Design body). E.g., creating some mirror constraint across a construction line or making some tangent constraints or even deleting some elements.
E.g., deleting `Base.Sketch006.ExternalEdge2` in this:
[PowerRelayHolder.FCStd.zip](https://github.com/realthunder/FreeCAD/files/13814591/PowerRelayHolder.FCStd.zip)
Or select `Edge3` and `Edge4` in `Base.Sketch008` and select the tangent constraint in this:
[PowerRelayHolder2.FCStd.zip](https://github.com/realthunder/FreeCAD/files/13814678/PowerRelayHolder2.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kevenwyld, 2024-01-03:
I'm also experiencing frequent crashes (segfaults) with mostly coincident and symmetry constraints in sketcher.  A stack trace in the console from one of them is below.

```
OS: Ubuntu Core 20 (sway/sway)
Word size of FreeCAD: 64-bit
Version: 2023.1231.0.38924 (Git) Snap 122
Build type: Release
Branch: tag: 20231231tip
Hash: 7471b6a2720ee4973b9477a55713070b056c42e8
Python 3.8.10, Qt 5.15.7, Coin 4.0.1, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Curves 0.6.15
  * fasteners 0.4.71
```

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f4def1cb090]
#1  0x7f4ce66fd7f0 in SketcherGui::ViewProviderSketch::drawConstraintIcons() from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x7e0
#2  0x7f4ce66fe3e1 in SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x241
#3  0x7f4df27d6ea4 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x34
#4  0x7f4df23ee25f in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x38f
#5  0x7f4df27d6f56 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x76
#6  0x7f4df23ee25f in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x38f
#7  0x7f4df27d84bc in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x48c
#8  0x7f4df27d9745 in Gui::SelectionSingleton::clearCompleteSelection(bool) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x485
#9  0x7f4df27d9ede in Gui::SelectionSingleton::clearSelection(char const*, bool) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x54e
#10  0x7f4ce66c0d5d in SketcherGui::TaskSketcherElements::on_elementsWidget_itemSelectionChanged() from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x11d
#11  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ec654) [0x7f4df008d654]
#12  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x466a86) [0x7f4df0edba86]
#13  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x46c2c9) [0x7f4df0ee12c9]
#14  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ec654) [0x7f4df008d654]
#15  0x7f4df0006f74 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x44
#16  0x7f4df000c6f8 in QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x4e8
#17  0x7f4df000fd62 in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x212
#18  0x7f4df000934b in QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x19b
#19  0x7f4df0e56e1d in QAbstractItemView::setCurrentIndex(QModelIndex const&) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x9d
#20  0x7f4ce66c2ac7 in SketcherGui::TaskSketcherElements::slotElementsChanged() from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x4b7
#21  0x7f4df2182b80 in boost::signals2::detail::signal_impl<void (), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::operator()() from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x360
#22  0x7f4ce762981c in Sketcher::SketchObject::acceptGeometry() from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0x5c
#23  0x7f4ce76432eb in Sketcher::SketchObject::onChanged(App::Property const*) from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0xfab
#24  0x7f4df16abcc4 in App::Property::touch() from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x124
#25  0x7f4df16aa51c in App::Property::hasSetValue() from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x6c
#26  0x7f4de8f58b65 in Part::PropertyGeometryList::setValues(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> >&&) from /snap/freecad-realthunder/122/usr/lib/Part.so+0x2d5
#27  0x7f4ce763be30 in Sketcher::SketchObject::solve(bool) from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0x560
[... cut ...]
#63  0x7f4df005cd54 in QCoreApplication::exec() from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#64  0x7f4df2047a24 in Gui::Application::runApplication() from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x404
#65  /snap/freecad-realthunder/122/usr/bin/FreeCADLink(+0xa380) [0x55ea54ed8380]
#66  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f4def1ac083]
#67  /snap/freecad-realthunder/122/usr/bin/FreeCADLink(+0xa79e) [0x55ea54ed879e]
```

GinTonic99, 2024-01-03:
I can confirm the crashes in sketcher with very simple operations in both the latest appimage and snap.

OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2024.101.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 97ccc0b4c01d671f5a85116ffa5f71464f1eee1c
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: German/Austria (de_AT)

`[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2023.1231.0.38924 (Git) Snap 122
Build type: Release
Branch: tag: 20231231tip
Hash: 7471b6a2720ee4973b9477a55713070b056c42e8
Python 3.8.10, Qt 5.15.7, Coin 4.0.1, Vtk 7.1.1, OCC 7.6.3
Locale: German/Austria (de_AT)
Installed mods: 
  * sheetmetal 0.3.13
  * POV-Ray-Rendering
  * Curves 0.6.21
  * fasteners 0.4.75
  * lattice2 1.0.0
  * QuickMeasure 2022.10.28
  * TabBar
  * Glass
  * Assembly4 0.50.6
  * FreeCAD_SketchArch
  * PieMenu 1.2.7 (Disabled)
  * woodworking 0.21.1.33694
  * CurvedShapes 1.0.5
  * fasteners.backup1702303153.0736592 0.4.70 (Disabled)
  * Render 2023.12.28
[/code]
`

here the terminal output in snap

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f112f7cb090]
#1  /lib/x86_64-linux-gnu/libc.so.6(+0x1886f5) [0x7f112f9106f5]
#2  0x7f1131da4cbd in Data::IndexedName::set(char const*, int, std::vector<char const*, std::allocator<char const*> > const&, bool) from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x4d
#3  0x7f103c97ca9e in Mesh::MeshObject::getSubElementAsMesh(char const*) const from /snap/freecad-realthunder/122/usr/lib/Mesh.so+0x3e
#4  0x7f103c988f6e in Mesh::Feature::getSubObject(char const*, _object**, Base::Matrix4D*, bool, int) const from /snap/freecad-realthunder/122/usr/lib/Mesh.so+0x9e
#5  /snap/freecad-realthunder/122/usr/lib/Part.so(+0x337801) [0x7f112957f801]
#6  0x7f11295826e5 in Part::Feature::getTopoShape(App::DocumentObject const*, char const*, bool, Base::Matrix4D*, App::DocumentObject**, bool, bool, bool) from /snap/freecad-realthunder/122/usr/lib/Part.so+0x415
#7  0x7f112976925e in Part::Module::getShape(Py::Tuple const&, Py::Dict const&) from /snap/freecad-realthunder/122/usr/lib/Part.so+0x1ee
#8  0x7f1129756153 in Part::Module::invoke_method_keyword(void*, Py::Tuple const&, Py::Dict const&) from /snap/freecad-realthunder/122/usr/lib/Part.so+0x43
#9  /snap/freecad-realthunder/122/usr/lib/libFreeCADBase.so(method_keyword_call_handler+0x35a) [0x7f11301fc47a]
#10  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8748) [0x7f112fe50748]
#11  0x7f1131cb1b62 in App::PropertyContainerPy::getCustomAttributes(char const*) const from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0xa72
#12  0x7f1131cb22b7 in App::PropertyContainerPy::_getattr(char const*) from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x17
#13  0x7f11302a27d4 in Base::PyObjectBase::__getattro(_object*, _object*) from /snap/freecad-realthunder/122/usr/lib/libFreeCADBase.so+0xa4
#14  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyObject_LookupAttr+0x3b) [0x7f112fe0675b]
#15  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x1cdd6d) [0x7f112fd75d6d]
#16  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x25f547) [0x7f112fe07547]
#17  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d) [0x7f112fc1cd6d]
#18  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x12fd) [0x7f112fc1e46d]
#19  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x8006b) [0x7f112fc2806b]
#20  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d) [0x7f112fc1cd6d]
#21  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xea8) [0x7f112fc1e018]
#22  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x8006b) [0x7f112fc2806b]
#23  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8d37) [0x7f112fe50d37]
#24  /snap/freecad-realthunder/122/usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyVectorcall_Call+0x60) [0x7f112fe50840]
#25  0x7f1131c10b3e in App::FeaturePythonImp::execute() from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x18e
#26  0x7f1131c12a3a in App::FeaturePythonT<App::GeoFeature>::execute() from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x1a
#27  0x7f1131b223c2 in App::DocumentObject::recompute() from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0xc2
[... 45 lines cut ...]
#73  0x7f11311d8e53 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#74  0x7f11311e1e57 in QApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x397
#75  0x7f11326ff628 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x98
#76  0x7f113064c0aa in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
#77  0x7f11311e1157 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-g [... cut ...]
#92  0x7f113064abab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#93  0x7f1130652d54 in QCoreApplication::exec() from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#94  0x7f113263da24 in Gui::Application::runApplication() from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x404
#95  /snap/freecad-realthunder/122/usr/bin/FreeCADLink(+0xa380) [0x558a90dc0380]
#96  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f112f7ac083]
#97  /snap/freecad-realthunder/122/usr/bin/FreeCADLink(+0xa79e) [0x558a90dc079e]

mzient, 2024-01-03:
I can confirm that the newest release is barely usable due to crashes in sketcher. I could use the previous one for hours without any serious incidents.

m-sundman, 2024-01-04:
@realthunder Are those `edit->timer.start(100)` introducing race conditions?

realthunder, 2024-01-04:
Why would that be? The crash happens because of de-syncrhonization of the sketch geometry and its visual representation. The timer is meant to delay visual update in (potentially recursive) UI event handling.

closed 2024-01-04 by realthunder

m-sundman, 2024-01-04:
> Why would that be? [...] The timer is meant to delay visual update [...]

It just looks like it, although I'm not familiar with the code. Anything that relies on timing (which is what "delay" presumably implies) could cause a race condition. What if the scheduler happens to not execute whatever other thread that is supposed to execute before the delayed thread? E.g., maybe there's a 150 ms scheduling interruption (say, because of page swapping), so that 100 ms delay is completely skipped. Does it then crash?
If you want 2 threads to execute in a specific order then maybe have one of them wait for a signal from the other?

realthunder, 2024-01-04:
> E.g., maybe there's a 150 ms scheduling interruption (say, because of page swapping), so that 100 ms delay is completely skipped. Does it then crash

This is indeed possible. But it is not a `race condition` because there is only single thread execution. This is a state inconsistency problem which exists long time in Sketcher code. Upstream Sketcher has a huge refactor which I only ported part of it. Not sure if they have fundamentally solved it. A proper solution would be to guard any `App` code access from `Gui` side and check for update. There is just too many such kind of access without guard in the original code before the refactor.

commit 69acefd1 (2024-01-04): Sketcher: fix crash on geometry change

Related #917
Fixes #922
commit e38e4646 (2024-05-28): Sketcher: fix crash on geometry change

Related #917
Fixes #922