realthunder/FreeCAD #924: [Problem] Incompatibility with CfdOF workbench

trilbytim, 2024-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.101.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 97ccc0b4c01d671f5a85116ffa5f71464f1eee1c
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CfdOF 1.24.9
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

I have the CfdOF workbench installed, it runs fine with standard FreeCAD 0.21.1, but not with Link Branch, which gives an error `During initialization the error "No module named 'CfdOF.compat'" occurred in /home/mt1tjs/.local/share/FreeCAD/Mod/CfdOF/./InitGui.py` and also will not open the workbench, giving the same error. I have attached log files from opening the standard version and the Link Branch (both using AppImages) 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-01-06:
Please update CfdOF workbench and see if it works. It works fine on my computer.

trilbytim, 2024-01-08:
I'm on version 1.24.9 of the CfdOF workbench. According to the message in Addon Manager this is the latest version available so I can't update it. I have tried uninstalling and reinstalling the workbench, but this hasn't solved it.

closed 2024-01-08 by realthunder