realthunder/FreeCAD #925: [Problem] Failed to create OpenGL context

nicocesar, 2024-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
this is freecad-realthunder (beta) 2023-12-31-g7471b6a2 

I can't open the "About dialog" since  it crashes before showing main screen
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

When I install and start freecad-realthunder from scratch seems that iris drivers arent available:

```
nico@automata:~$ snap install --beta freecad-realthunder
2024-01-03T16:47:24-05:00 INFO cannot update "cups", will not have required content "foo": snap has no updates available
freecad-realthunder (beta) 2023-12-31-g7471b6a2 from ZHENG LEI (realthunder) installed
nico@automata:~$ freecad-realthunder
FreeCAD 0.21.0, Libs: 2023.1231.0R38924 (Git)
Adding snap-specific PYTHONPATH to sys.path: /home/nico/snap/freecad-realthunder/common/.local/lib/python3.8/site-packages:/snap/freecad-realthunder/122/lib/python3.8/site-packages:/snap/freecad-realthunder/122/usr/lib/python3/dist-packages
MESA: warning: Driver does not support the 0xa7a0 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa7a0 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Qt: Session management error: None of the authentication protocols specified are supported
0.931978 <asm3.main> init_gui.py(14): no solver backend found
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)
```

But the drivers seems to be working:

```
nico@automata:~$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-P [Iris Xe Graphics] (rev 04)
nico@automata:~$ glxinfo
name of display: :1
display: :1  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_get_drawable_type, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_ATI_pixel_format_float, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_NV_float_buffer, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
```

and the version via APT does work:

```
nico@automata:~$ apt-cache policy freecad
freecad:
  Installed: 0.20.2+dfsg1-9
  Candidate: 0.20.2+dfsg1-9
  Version table:
 *** 0.20.2+dfsg1-9 500
        500 http://apt.pop-os.org/ubuntu mantic/universe amd64 Packages
        500 http://apt.pop-os.org/ubuntu mantic/universe i386 Packages
        100 /var/lib/dpkg/status
```

But I need 0.21 !!

I Cant find this problem reported anywhere else. IF anybody has pointers to a solution copy the links int the comments. Thanks

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nicocesar, 2024-01-07:
Temporary solution for me:

- go to https://github.com/realthunder/FreeCAD/releases and choose AppImage
- download FreeCAD-Link-Stable-Linux-x86_64-py3.11-20240104.AppImage # for example
- chmod +x FreeCAD-Link-Stable-Linux-x86_64-py3.11-20240104.AppImage
- LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./FreeCAD-Link-Stable-Linux-x86_64-py3.11-20240104.AppImage

realthunder, 2024-02-03:
Could you please try the latest stable snap release and see if it is fixed?

nicocesar, 2024-02-05:
it works now, I'm using beta: 

```
$ snap install --beta freecad-realthunder
freecad-realthunder (beta) 2023-12-31-g7471b6a2 from ZHENG LEI (realthunder) installed
$ freecad-realthunder
FreeCAD 0.21.0, Libs: 2023.1231.0R38924 (Git)
(....)
```

nicocesar, 2024-02-05:
Either way, this is working fine. we can close the ticket