realthunder/FreeCAD #930: The MacOS version of Link Branch functions differently than the Windows version.[Problem]

jasonbrowne44, 2024-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 12.6
Word size of FreeCAD: 64-bit
Version: 2024.104.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 975453f6aa33093edbcc835760d7a65123059b98
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Problem description

When I attempt to build a simple electronic enclosure, I am unable to complete it under MacOS.  On Widows 10, the experience is much improved.  I receive a number of recompute errors and this means that many surfaces disappear randomly.  I attempt to start over, but I end up exactly in the same place.  Maybe there is an obvious reason for this, but I don't know if there are any differences between the code on the two platforms.  Would it be worth it to walk through the two platforms with my example 3D drawing to see why they are so different?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-01-13:
Could you please demonstrate the differences using some simple model?

jasonbrowne44, 2024-01-13:
I’ll try to get it down as simple as possible.

Thanks,

Jason

jasonbrowne44, 2024-01-14:
[Simple Drawing Not Showing in MacOS.FCStd.zip](https://github.com/realthunder/FreeCAD/files/13930390/Simple.Drawing.Not.Showing.in.MacOS.FCStd.zip)

This should be a simple cube.  Let me know if it looks correct on your MacOS machine.