realthunder/FreeCAD #932: [Problem] symmetrical constraint not working anymore

not-a-number-dot-org, 2024-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 2024.113.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 4235ee9c92d9822b434ec2bc8da7cee729fd9f3b
Python 3.11.7, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Adding a symmetrical constraints result in the constraint being tagged Redundant in many situations.
See image for example which is quite a simple example.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

not-a-number-dot-org, 2024-01-14:
Maybe related to arcs and not to symmetrical constraint.
[image]

not-a-number-dot-org, 2024-01-14:
This may be related to the use of arcs and not to the use of symmetrical constraints
[image]

realthunder, 2024-01-15:
It is just a warning. There is indeed a redundant constraint in your OP, which is the point coincidence. The symmetric to X axis will constraint the arc center on to X axis. So, instead of a point coincidence, you can replace it with a point on Y axis constraint.

The enhanced warning message is most likely due to upstream changes in Sketcher solver

closed 2024-01-21 by not-a-number-dot-org

oerkel47, 2024-03-22:
There is still funny behaviour going on with this. I am not sure if this is really working as intendend.

https://github.com/realthunder/FreeCAD/assets/57231822/7b1c205e-b2dc-49ad-ad2c-72657b04319e