realthunder/FreeCAD #934: Start page options in preferences not working

CafedoCampo, 2024-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 2024.114.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 4235ee9c92d9822b434ec2bc8da7cee729fd9f3b
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Preferences for Start page options seem to not be working.  There are 3 options in the preferences interface: (1) switch workbench after loading; (2) close start page after loading; and (3) close and switch on opening file.  None of them work. 

Also, see comment below.

### Anything else?

In case it is relevant, there is a path issue when I install the daily updates on my mac.  This may be specific to mac.  I get a “no solver backend found” error with each update.  My work-around is to edit the sys.slvs.py file to include the path in sys.path:

Location of file:
_‘/Applications/FreeCAD.app/Contents/Resources/Mod/asm3/freecad/asm3/sys.slsvs.py’_

Lines added:
_import sys
sys.path.append('/opt/homebrew/lib/python3.11/site-packages/py_slvs')_

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-01-22:
Hi, I just tested on MacOS, but cannot reproduce the problem. Note that these options applies to what will happen when you open/create a file.

I'll deal with solver not found problem later. In the future, please create separate issue for each problem.

CafedoCampo, 2024-01-22:
Thanks very much for testing.  I'll keep playing with the Start page issue to see if I can figure out what is happening, but will close for now. 

It sounds like you have the solver issue covered, but please let me know if you want me to add it as a new issue.