realthunder/FreeCAD #943: Sketch problems with last LinkMerge

wsteffe, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.38946 (Git)
Build type: Unknown
Branch: LinkStable
Hash: 2eaff092937eeee5c382f9ddba0472c86c8b80d1
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * asm3 0.12.0
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

In the image below I added a small rectangle to thew first sketch and I trimmed a couple of edges of this rectangle. FC produced the following error:

> Sketch> SketchObject.cpp(11357): cannot find subname ;g86.edge55

After that it was not possible to move the remaining two edges of added rectangle, In this case FC produced following additional errors:

> 12:00:33  Unhandled unknown C++ exception in ViewProvider::eventCallback (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)
> 12:05:54  App.getDocument('planimetria').getObject('Sketch').movePoint(55,0,App.Vector(-264.600341,-11.199707,0),1)
> 12:05:54  Drag curve: Not able to move point with the id and type: (55, 0)
> 
> 

[planimetria.zip](https://github.com/realthunder/FreeCAD/files/14075730/planimetria.zip)

 
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-09:
> Sketch> SketchObject.cpp(11357): cannot find subname ;g86.edge55

This is a harmless error which I'll fix shortly. The problem probably happens when you drag the curve somehow lock up the sketcher solver. Is this problem reproducable? I can't seem to do that.

wsteffe, 2024-03-09:
I encontered very often this kind of problem in the last year.  I think it started with the last upstream sync in may 2023.
Actulally it is not necessary to trim to reproduce this problem. 
Ypu may see it deleting the vertical length highlighted in the following picture:

[image]

and then trying to move the vertical position of the vertex highlighted in the next picture:

[image]

realthunder, 2024-03-10:
Still can't reproduce the problem. Could you please post the file in the state just before dragging? And also please post your user configuration file.

wsteffe, 2024-03-10:
Folowing test file before dragging, user.cfg and an additional image. The problem arise with a vertical drag of the highligted vertex. It happens also when I drag the other vertex belonging to the same horizontal edge. 
 
[planimetria_test.zip](https://github.com/realthunder/FreeCAD/files/14549013/planimetria_test.zip)
[user.zip](https://github.com/realthunder/FreeCAD/files/14549017/user.zip)

[image]

realthunder, 2024-03-16:
It is indeed caused by your settings. More specifically your sketcher solver setting as shown below. Mine is 100, which is the default value, I think.

[image]

wsteffe, 2024-03-16:
Thanks for spotting the problem. 
But it is quite strange to me that a smaller number of solver iteration prevented dragging a vertex.

closed 2024-03-16 by wsteffe