realthunder/FreeCAD #946: Input characters can't be changed by IBUS in Snap release

ftdt10, 2024-01-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.124.0.38947 (Git) Snap 127
Build type: Release
Branch: tag: 20240123stable
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Input character change for CJKV language doesn't work only on the Snap release using IBUS.
(I think is is related to https://forum.snapcraft.io/t/compatibility-with-cjkv-input-method-frameworks/7788/1)

I tested it on the Appimage release, but input language can be changed normally in there.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ftdt10, 2024-02-15:
More test report.

If the HOME environmental variable for Qt is set when launching AppImage release, Input language can't be changed.
If the HOME environmental variable for Qt is not set when launching AppImage release, Input language can be changed.

https://wiki.freecad.org/Download#Notes_for_GNU.2FLinux_users
https://wiki.freecad.org/Start_up_and_Configuration#HOME

realthunder, 2024-03-23:
Please check with the latest release

ftdt10, 2024-03-25:
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.322.0.38974 (Git) Snap 130
Build type: Release
Branch: tag: 20240322stable
Hash: da98001cfe5cc50cb2e9e4c241626d07dd4b06f5
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)

I tested it, but it still doen't work in the Snap release.

On Ubuntu 23.10, configured like below

Settings>Region&Language>ManageInstalledLanguages,
'Install/Remove Languages" - add for CJK languages
'Keyboard input method system' - change to IBus

Settings>Keyboard>InputSources,
'Add an Input Source' - add for CJK languages

realthunder, 2024-04-08:
Still can't get it to work for ibus. You can try fcitx though.

ftdt10, 2024-04-09:
Fcitx5 works. Input language can be changed now in the Snap release. Removed Ibus.
Thanks for the fix.

PS: Regarding the program crash, the Snap release seems to work more stably than AppImage.

[code]
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.409.0.38991 (Git) Snap 132
Build type: Release
Branch: tag: 20240407stable
Hash:973e4821bcd19a5a1af9ae3e1d2ed961aad0a8ab
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.2
Locale: English/United States (en_US)
[/code]

closed 2024-04-09 by ftdt10