realthunder/FreeCAD #947: "Size of toolbar icons" setting doesn't work

ftdt10, 2024-02-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.124.0.38947 (Git) Snap 127
Build type: Release
Branch: tag: 20240123stable
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Problem description

The "Size of toolbar icons" setting at menu-general-preferences doesn't work properly in the following case.

Steps to reproduce
1. Change icons size value by menu-Tools-EditParameter-BaseApp-Preferences-General-ToolbarIconSize
2. Switch between workbenches, then the Icons size is not same as the setting value.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-09:
> 2\. Switch between workbenches, then the Icons size is not same as the setting value.

Could you please provide a screencast for this behavior? I can't reproduce it.

ftdt10, 2024-03-09:
Other steps to reproduce it.
0. run the Snap release FreeCAD 
1. change workbench to Assembly3, for example
2. set "Size of toolbar icons" to 'Small(16px)' by menu-Edit-Preferences-General, then clk OK
3. set ToolbarIconSize to 11 by menu-Tools-EditParameter-BaseApp-Preferences-General, then clk Close
4. go to again Menu-Edit-Preferences-General-"Size of toolbar icons", and change between 'Small(16px)' and 'Custom(11px)' several times
then, see the icons at top toolbars. The size of icon sizes in each toolbar is irregular.

commit 53befa09 (2024-03-10): Gui: fix customized main window toolbar size handling

Fixes #947
closed 2024-03-10 by realthunder