realthunder/FreeCAD #948: Shadow lighting direction is reset incorrectly

ftdt10, 2024-02-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.124.0.38947 (Git) Snap 127
Build type: Release
Branch: tag: 20240123stable
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.10.12, Qt 5.15.10, Coin 4.0.1rt, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Problem description

if setting is changed in DocumentRoot-ViewTab-Shadow, lighting direction resets incorrectly.
And if the document file is closed and reopened, the lighting direction will be reset again.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-09:
Could you please provide some picture showing which parameters you've changed, and that changed before/after document reopened?

ftdt10, 2024-03-09:
More detail steps to reproduce

1. run "memu-View-DrawStyle-Shadow", and run it again to show the directional light icon.
2. change the direction of light by dragging the arrow
3. change "Light Intensity" to other value, for example
then the direction of the light is reset

commit 4af307db (2024-03-10): Gui: fix shadow light property sync problem with dragger

Fixes #948
closed 2024-03-10 by realthunder