realthunder/FreeCAD #951: Realthunder crashes from redundant constraints in sketch

HomicidalBrainiac, 2024-02-04:
Entire software crashes with segmentation fault when introducing redundant constraints to a sketch.

OS: Ubuntu Core 20
Word size of FreeCAD: 64-bit
Version: 2023.1231.0.38924 (Git) Snap 122
Build type: Release
Branch: tag: 20231231tip
Hash: 7471b6a2720ee4973b9477a55713070b056c42e8
Python 3.8.10, Qt 5.15.7, Coin 4.0.1, Vtk 7.1.1, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x70a973168090]
realthunder/FreeCAD_assembly3#1  /lib/x86_64-linux-gnu/libstdc++.so.6(__dynamic_cast+0x1f) [0x70a9733dcddf]
realthunder/FreeCAD_assembly3#2  0x70a8f20ff166 in SketcherGui::ViewProviderSketch::drawConstraintIcons() from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x156
realthunder/FreeCAD_assembly3#3  0x70a8f21003e1 in SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x241
realthunder/FreeCAD_assembly3#4  0x70a97674dea4 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x34
realthunder/FreeCAD_assembly3#5  0x70a97636525f in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x38f
realthunder/FreeCAD_assembly3#6  0x70a97674df56 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x76
realthunder/FreeCAD_assembly3#7  0x70a97636525f in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x38f
realthunder/FreeCAD_assembly3#8  0x70a97674f4bc in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x48c
realthunder/FreeCAD_assembly3#9  0x70a976750745 in Gui::SelectionSingleton::clearCompleteSelection(bool) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x485
realthunder/FreeCAD_assembly3#10  0x70a976750ede in Gui::SelectionSingleton::clearSelection(char const*, bool) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x54e
realthunder/FreeCAD_assembly3#11  0x70a8f20c2d5d in SketcherGui::TaskSketcherElements::on_elementsWidget_itemSelectionChanged() from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x11d
realthunder/FreeCAD_assembly3#12  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ec654) [0x70a974004654]
realthunder/FreeCAD_assembly3#13  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x466a86) [0x70a974e52a86]
realthunder/FreeCAD_assembly3#14  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x46c2c9) [0x70a974e582c9]
realthunder/FreeCAD_assembly3#15  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ec654) [0x70a974004654]
realthunder/FreeCAD_assembly3#16  0x70a973f7df74 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x44
realthunder/FreeCAD_assembly3#17  0x70a973f836f8 in QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x4e8
realthunder/FreeCAD_assembly3#18  0x70a973f86d62 in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x212
realthunder/FreeCAD#460  0x70a973f8034b in QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x19b
realthunder/FreeCAD_assembly3#20  0x70a974dcde1d in QAbstractItemView::setCurrentIndex(QModelIndex const&) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x9d
realthunder/FreeCAD_assembly3#21  0x70a8f20c4ac7 in SketcherGui::TaskSketcherElements::slotElementsChanged() from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x4b7
realthunder/FreeCAD_assembly3#22  0x70a9760f9b80 in boost::signals2::detail::signal_impl<void (), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::operator()() from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x360
realthunder/FreeCAD_assembly3#23  0x70a8f302b81c in Sketcher::SketchObject::acceptGeometry() from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0x5c
realthunder/FreeCAD_assembly3#24  0x70a8f30452eb in Sketcher::SketchObject::onChanged(App::Property const*) from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0xfab
realthunder/FreeCAD_assembly3#25  0x70a975622cc4 in App::Property::touch() from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x124
realthunder/FreeCAD_assembly3#26  0x70a97562151c in App::Property::hasSetValue() from /snap/freecad-realthunder/122/usr/lib/libFreeCADApp.so+0x6c
realthunder/FreeCAD_assembly3#27  0x70a96d075b65 in Part::PropertyGeometryList::setValues(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> >&&) from /snap/freecad-realthunder/122/usr/lib/Part.so+0x2d5
realthunder/FreeCAD_assembly3#28  0x70a8f303de30 in Sketcher::SketchObject::solve(bool) from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0x560
realthunder/FreeCAD_assembly3#29  0x70a8f304323d in Sketcher::SketchObject::delConstraints(std::vector<int, std::allocator<int> >, bool) from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0x40d
realthunder/FreeCAD_assembly3#30  0x70a8f3043b92 in Sketcher::SketchObject::autoRemoveRedundants(bool) from /snap/freecad-realthunder/122/usr/lib/Sketcher.so+0x172
realthunder/FreeCAD_assembly3#31  0x70a8f201ccc6 in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*, bool&) from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0xc6
realthunder/FreeCAD_assembly3#32  0x70a8f201cd12 in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*) from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x22
realthunder/FreeCAD_assembly3#33  0x70a8f2066735 in CmdSketcherConstrainSymmetric::activated(int) from /snap/freecad-realthunder/122/usr/lib/SketcherGui.so+0x835
realthunder/FreeCAD_assembly3#34  0x70a9760f2a48 in Gui::Command::_invoke(int, bool) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x258
realthunder/FreeCAD_assembly3#35  0x70a9760f2f15 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x155
realthunder/FreeCAD_assembly3#36  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ec622) [0x70a974004622]
realthunder/FreeCAD_assembly3#37  0x70a974b53996 in QAction::triggered(bool) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
realthunder/FreeCAD_assembly3#38  0x70a974b56048 in QAction::activate(QAction::ActionEvent) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa8
realthunder/FreeCAD_assembly3#39  /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x267d00) [0x70a974c53d00]
realthunder/FreeCAD_assembly3#40  0x70a974c53f25 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe5
realthunder/FreeCAD_assembly3#41  0x70a974d4dc0e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
[... cut ...]
realthunder/FreeCAD_assembly3#64  0x70a973fd3d54 in QCoreApplication::exec() from /snap/freecad-realthunder/122/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
realthunder/FreeCAD_assembly3#65  0x70a975fbea24 in Gui::Application::runApplication() from /snap/freecad-realthunder/122/usr/lib/libFreeCADGui.so+0x404
realthunder/FreeCAD_assembly3#66  /snap/freecad-realthunder/122/usr/bin/FreeCADLink(+0xa380) [0x647227c42380]
realthunder/FreeCAD_assembly3#67  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x70a973149083]
realthunder/FreeCAD_assembly3#68  /snap/freecad-realthunder/122/usr/bin/FreeCADLink(+0xa79e) [0x647227c4279e]

realthunder, 2024-02-04:
Could you please update to snap stable version? You need to manually do that because of change of channel to stable. This crash is fixed in abf8bb77dcb.

HomicidalBrainiac, 2024-02-05:
Wow thanks for your patient and kind response! Indeed I was unaware of the change.  Go ahead and close this.I want to sincerely thank you and anyone else contributing to realthunder. The world needed this extra step so badly and you all truly did a good thing!  Thank you so much!!!

closed 2024-03-23 by realthunder