realthunder/FreeCAD #954: [Problem]

gneiss15, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 18.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2024.123.0.14555 (Git shallow) AppImage
Build type: Release
Branch: HEAD
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * ThreadProfile 1.89.0
  * Manipulator 1.5.7
  * InventorLoader 1.4.0
  * Assembly4 0.50.7
  * nurbs
  * ExplodedAssembly
  * sheetmetal 0.4.0
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

There are some "path-problems" when using the AppImage-Version of FreeCad (I believe, that applies not only to the Link-Branch).
The path of/for some workbenches are stored inside "user.cfg".
This path is something like "/tmp/.mount_*" when using the AppImage.
This path changes on every start of the AppImage. 
The Problem ocurs (only examples):
- inside Techdraw: Patterns, LineGroups, Template & Symbols are not found on next open of document after creating the item
- AddOnManager: Path to (local) pip & git

To overcome this, the path must be adjusted on every start of the app.
I have done this by modifing AppRun.
I simply add the code below before the lines:
```
cd $PREFIX/bin
"./$name" "$@"
```

I hope this is helpfull.
Maybe the code must be adjusted a bit to be "universal". For me is's OK.

```
###############################################################
# Adjust path inside "${USER_CFG}" to current mount path
###############################################################

while getopts ":u:" option; do
  case $option in
    u) USER_CFG="$OPTARG";;
    *) ;;
  esac
done

if [ -z ${USER_CFG} ]; then
  USER_CFG="~/.config/FreeCAD/link.user.cfg"
fi

sed -i s:/tmp/\.mount_[^/]*/:${HERE}/:g ${USER_CFG}
```

### Anything else?

A similar problem occurs when using the AppImage together with the sketches on a USB stick:
Recent files, etc. are not updated to the new path to the AppImage file (the USB stick).
This can be adjusted by launching Freecad via a shell script that adjust these paths.
I use one myself. If you are interested I can deliver it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gneiss15, 2024-02-14:
I would mention another "path-problem" when using Draft->ShapeString. Here the path to the font-file is placed into the sketch.file (.FCStd). So when You modify anything later (for ex move the position) You will get an Error " Can not open font file"..
This can only be fixed by a script that runs whenever a file was opened.
How to do this is (at the moment) out of my knwoledge...

realthunder, 2024-03-09:
AppImage can access your local computer file. So why not choose a file on your local file system?

gneiss15, 2024-03-11:
@realthunder 
Maybe I didn't describe the problem clearly enough.
The most important keyword is "PORTABILTY".
All files and programs are on a USB stick. This means, that I can work on (almost) any PC. My stick is bootable, too.
Using a file "on a local file system" is just the opposite of portability!

OK, I try again to explain what the problems are, but first some remarks:
1. If FreeCad uses relative instead of absolute paths these problems will be much lesser.
2. Even when not using an AppImage, but an installation of FreeCad on a stick problem 1 & 3 will arise.
3. A much better solution will be if FreeCad uses something like a “base” for the different “places” (like $HOME in Linux for files under the users home dir). Here it may use e.g. APPHOME, DRAWINGSHOME… Setting these environment vars before starting FreeCad may solve a lot of the mentioned problems.
4. The use of the path to the font-file in Draft→ShapeString is IMHO a Bug. Doing so makes the drawing "not portable". When I send such a drawing to somebody else, who didn’t has the used font, the drawing can’t be modified (by this user). At least if no changes to the text itself are made, the drawing should be editable (maybe by storing the text as vectors). I hope it is clear what I mean.
As long as I use only fonts which are located inside the FreeCad tree the same solution as for point 1 & 2 below can be used, but the "path replace" must be done “on drawing open”.

I identify three kind (places) of such "path-problems":
absolute paths containing (starting with) the mount point of ...
1. ... a removable media (found inside user.cfg)
2. ... the tmp-mount point of the AppImage (found inside user.cfg)
3. ... the tmp-mount point of the AppImage and/or the mount point of a removable media (found inside drawing/.FCStd files)

My current solution for 1 & 2 is: Replace the "last" paths with the new one.

For Point 2 that can only be done after the AppImage has mounted itself. So doing the modification inside AppRun (as noted above) is the right way.
For Point 1 that can be done before starting the AppImage, because the mount point of the removable media is known at this time.

Code snipped for 1:
`# because my AppImage is inside a dir “/Portables/FreeCad”, replacing the “mount path” of the stick can be done so:

LAST_MOUNT_PATH=$(Escape "$(sed -e '/^.*<FCText Name=.*>.*\/Portables\/FreeCad\//!d;s:^.*<FCText Name=[^>]*>\(.*\)/Portables/.*:\1:g;q' ${USER_CFG})")

CURRENT_MOUNT_PATH=$(Escape "$(readlink -f -- ${APPDIR} | sed -e "s:\(.*\)/Portables/.*:\1:g")")
`