realthunder/FreeCAD #958: Add Intersection Defining feature results in duplicate defining geometry.

SamuelFarinas, 2024-02-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 2024.123.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.8.18, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

The Add Intersection Defining feature adds duplicate defining geometry after clicking an edge.

Steps to reproduce:
1. Start a sketch.
2. Enable "Add/toggle intersection defining" feature.
3. Click an edge of a feature.
4. Observe "Elements" section in the Tasks pane.
5. Note that there are two defining lines instead of one.

This bug is problematic when creating sketches where closed geometry is required to support pad, revolution, and other operations.

[image]

Thanks, I like how the project is turning out so far!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-02-20:
Could you please provide a sample file showing the problem?

SamuelFarinas, 2024-02-21:
[SketchWithDuplicateExternalLine.zip](https://github.com/realthunder/FreeCAD/files/14354283/SketchWithDuplicateExternalLine.zip)

Hi, thanks for taking a look into it! Apologies, when I tried creating a simpler file, I found the bug is more specific, involving the Gear workspace. I have attached the project file above, where you'll have to set the file extension back to .FCStd to open it via FreeCAD. (I couldn't upload the file otherwise).

The revised repro steps are:

1. Create a body. In the Gear workbench, create an involute gear.
2. Start a sketch on the top face of the gear.
3. Enable "Add/toggle intersection defining" feature.
4. Click an edge on the top face of the gear.
5. Observe "Elements" section in the Tasks pane.
6. Note that there are two defining lines instead of one.
7. Try deleting one of the lines and observe an odd error.
8. Select the sketch edge in the UX and delete it. Notice both items in the "Elements" section are removed.

~ Samuel

commit 48cff1a9 (2024-03-10): Sketcher: fix linear approximation of BSpline external edge

Fixes #958
closed 2024-03-10 by realthunder