realthunder/FreeCAD #964: [Problem] Access violation message while editing sketch

vovodroid, 2024-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 2024.123.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.11.7, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English
Installed mods: 
  * Curves 0.6.13
  * ThreadProfile 1.87.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

Recently I started to experience "Access violation" message. I can't pinpoint it to specific action or project.

[image]

How can I collect logs, etc?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

realthunder, 2024-03-16:
Probably related to #967. Could you please find the 'crash.dmp' file in your user directory (`c:\users\<your_username>\AppData\Roaming\FreeCAD`) and post it here?

vovodroid, 2024-03-16:
I don't see crash.dmp in \AppData\Roaming\FreeCAD. Should I enabled it creation?

vovodroid, 2024-03-21:
Still in 20240317stable, not related to deleting external geometry. Just random errors while editing sketch.

realthunder, 2024-03-21:
I played around in windows build but failed to reproduce the problem. Could you please post a file and describe the action you took to trigger to error?

vovodroid, 2024-03-21:
Different files, just editing sketches, nothing special. For example in this file.

[500.zip](https://github.com/realthunder/FreeCAD/files/14696362/500.zip)

How can I enable logs/crash dump?

realthunder, 2024-03-22:
`Access violation` means some unknown nasty exception got caught by the application preventing a hard crash, so there is no dump file. I tried your file and still can't reproduce the problem. Could you please post your user settings here, Tools -> Edit Parameters -> Export'

vovodroid, 2024-03-22:
Is there any option for logs?

[setting.zip](https://github.com/realthunder/FreeCAD/files/14718945/setting.zip)