realthunder/FreeCAD #966: Step file exported from revolution shape is broken

wsteffe, 2024-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.38957 (Git)
Build type: Unknown
Branch: LinkStable
Hash: fef2b22db6aa7f6304a81a9fddb27a44b613abfc
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * asm3 0.12.0
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

The generated geometry associated with annexed FC document  seems good inside of FC:

[image]

But after exporting it in a step file and reimporting in Catia il looks like the folowing:
[image]

[filterV7.zip](https://github.com/realthunder/FreeCAD/files/14508645/filterV7.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wsteffe, 2024-03-07:
Thinking the problem could be related with the seem edge, now I have tried to make smaller rotation (90 deg instead of 360) but the problem persists:

 
[image]
[image]

Even if it happen in the opposite direction (step export instead of step import), I suspect that it is related with OCC issue https://tracker.dev.opencascade.org/view.php?id=33620

wsteffe, 2024-03-07:
Here is the FC document used for 90 deg rotation
[filterV7.zip](https://github.com/realthunder/FreeCAD/files/14523909/filterV7.zip)

realthunder, 2024-03-16:
Have you tried open the STEP file in FreeCAD? It looks normal to me. I also tried viewing the file in some online viewer and looks normal too. Maybe it's Cartia having problem? Could you please post your STEP file here?

wsteffe, 2024-03-16:
The step file is : [CMP_FILTER_V7_step.zip](https://github.com/realthunder/FreeCAD/files/14623681/CMP_FILTER_V7_step.zip)
I may confirm that it looks ok inside of FreeCAD. But it is not a Catia problem. 
I have opened it also with ZW3D and it looks bad as in Catia.

You may look also at case https://tracker.dev.opencascade.org/view.php?id=33618.
The original case posted by ebknudsen goes in the opposite direction but I thinks it is just the other face of the same problem. Using a commercial CAD (OneShape)  he revolved a profile made of circles arcs (while I had used ellipse arcs).  The exported/reimported step file looks good in OneShape but it looks bad in his viewer tool (not tried  with FreCAD) which is based on OpenCascade library. In my opinion OCC is doing something wrong when it exports this kind of structure (the file is not compliant with the step standard).